CR6Community/Marlin #168: [FR] Please facilitate cleaning the bed of "drool" at the end of the ABL routine

Thinkersbluff, 2021-02-01:
While the CF Devs recommend that the nozzle temp be kept lower than printing temperature while probing, some users prefer to perform ABL at their intended print temperatures.  Doing so, however, may result in the probe leaving small filament deposits on the bed at various probe points.

In the current design, the ABL routine parks the head at the front of the bed.  This makes it difficult for users to clean those deposits off the bed.

It may be helpful if the head was instead parked at the left rear corner, presenting clear access to the bed.
Alternatively, one might disable steppers after parking the head, allowing users to pull the bed forward for cleaning.

Sebazzz, 2021-02-04:
Centering the bed, but Z not too high.

closed 2021-02-04 by Sebazzz

Thinkersbluff, 2021-02-04:
I appreciate you including a fix in the next release.  I had hoped for clear access to the bed, though, and parked in the middle is different but not better for clearing bits left at the back row.

Leaving the head at the front left probably made Z Offset Homing a tad faster, right?  Maybe this fix is even a step backwards for some.  Parking the head anywhere is still going to leave the steppers locked, come to think of it, so probably not a good suggestion on my part.  

Users who prefer to use ABL and Z Offset at high temps could select Back->Prepare->Disable Steppers function to do this cleaning before proceeding to print.  We know ABL does not leave any blits at the Home position, so Z Offset is probably not at risk. (Unless you park the head at the center point after ABL :P  

If there was a Disable Steppers button on the Level Screen, it might be appreciated by some?

Thinkersbluff, 2021-02-06:
Intriguingly, when I view mesh values in my current build from 6 Feb 2021 extui, the head goes back to the middle of the bed after a few seconds of viewing time, though It does stay at the x=0,y=0 corner until I view mesh...