CR6Community/Marlin #171: If the SD card is write protect the printer won't start.

ckhmer1, 2021-02-04:
### Description

If the SD card is write protect the printer won't start.

### Steps to Reproduce

1. Put the SD card slider on lock, in order to write-protect it.
2. Insert the SD card on the printer
3. Switch on the printer

**Expected behavior:** 

The printer should startup, and maybe write an error message, SD Card write protect enabled

**Actual behavior:** 

Printer won't start.
Octoprint won't connect, showing an error message that the printer is restarting continuously.

**Logging:**

Impossible to get logs

### Additional Information

Mainboard 4.5.2.
Firmware Beta5 downloaded from the repo (Not compiled by myself)

ztakis, 2021-02-04:
Just tested this and it doesn't happen to my machine. Rel6-Pre2 / 4.5.2 board

Sebazzz, 2021-02-04:
This sounds as a possible issue local to your board, perhaps even to the bootloader.

closed 2021-02-04 by Sebazzz

ckhmer1, 2021-02-06:
Hi,
I'm sorry after more investigation seems that the SD card was not write-protect but faulty.
If the SD card is faulty, the printer is rebooting.
I don't know if this can be avoided.
Thanks for your great job.

Sebazzz, 2021-02-06:
That is _sort of_ intentional. A flaky SD card can cause Marlin to freeze. You don't want that to happen in the middle of a print.

ckhmer1, 2021-02-07:
You are right. But if you could avoid restarting marlin indefinitely and show a message, the SD card Is faulty, maybe helps.