CR6Community/Marlin #172: Can't home Z-axis after restart when high up

robeastwood, 2021-02-04:
### Description

If the printer is powered down while the z-axis gantry is still high up (say after completing a tall print) then when the power is restored, the home command no longer brings the gantry down far enough to touch the bed.

Note: I am using Octoprint. I am not sure if this is also an issue using native printing but i would suspect so.

### Steps to Reproduce

1. Perform a print at least ~200mm tall.
2. Hard Power off the machine
3. Power on the machine.
4. Home

**Expected behavior:** 

X, Y and Z axis to home.

**Actual behavior:** 

X & Y axis home, Gantry is left ~190mm? high.
Next print prints in mid-air.
Cannot move head down, it thinks it's at 0mm already.

[image]

The only solution I have at the moment is to disable steppers and manually wind the screws down to the point where it will auto home again.

**Logging:**

Send: G91
Recv: ok
Send: G28 X0 Y0
Recv: SetNewScreen: 61
Recv: SetNewScreen (saving): 61
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: SetNewScreen: 43
Recv: X:-5.00 Y:-2.00 Z:15.00 E:0.00 Count X:-400 Y:-160 Z:6000
Recv: Waiting to heat-up again before continueing
Recv: //action:notification Waiting for heat-up...
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: //action:notification CR-6 Community Ready.
Recv: ok
Send: G90
Recv: ok
[...]

### Additional Information

Standard hardware, v4.5.2 board

Firmware: CR6Comm-CR6-Rel5
https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip

Sebazzz, 2021-02-04:
Please try it with the [prerelease 6 pre2](https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.3-cr6-community-release-6-pre2) to see what the situation is on the latest firmware. I'd also like a picture of you hotend/gantry to see if you've done anything odd influencing homing.

If you still have the issue:
- Provide a minimal reproducible gcode file
- Check if the optical sensor on the gantry is lit. To me it sounds like it is malfunctioning because the Z-probe isn't supposed to activate until you're in the "zone.

robeastwood, 2021-02-04:
I will try prerelease6 when i get a chance, but I don't have time tonight to upgrade, and i am also no g-code wizard so im not sure how to do that sorry!

I can get more pictures/videos as you require if you can explain what would be useful to see?

But, the gantry/hotend is completely standard, and homing does work fine normally when the print head is nearer the bed.
I'm confident that there are no physical issues, z-probe strain guage is working, the optical sensor is working also - I can auto-mesh-level etc once it's been manually brought down to a point it seems happy with. And it does print perfectly fine when its done so.

Cheers!

robeastwood, 2021-02-04:
Ohe, the only thing thats not standard is that i've stuck a magnetic bed on the plate. But it behaved the same way before I did that so i dont think that makes it any different.

Thinkersbluff, 2021-02-07:
@robeastwood Did the printer even try to lower Z or did it just go up a bit and end the homing process there?

robeastwood, 2021-02-08:
@Thinkersbluff  I tried to test this and record it happening, but I was unable to replicate the issue.
I will monitor it next time I have to restart the printer and take better notes about the cause and behaviour.

closed 2021-02-08 by robeastwood