CR6Community/Marlin #174: [FR] Show Warning when Probed Mesh Min/Max range exceeds that which can be compensated for using ABL.

gleonard3, 2021-02-05:
It's not clear to me what that actual acceptable range is, but I have definitely experienced exceeding that limit, whereby some physical hardware tweaking of the printer itself is required.  If we worded a warning like this correctly, I think it would help a lot of new users.

Something like "WARNING:  Recommended Frame Squareness Check and Re-Level."

Sebazzz, 2021-02-05:
There isn't really a limit. Marlin is about to compensate for 10mm in difference just fine. Try blocking the nozzle at one of the leveling points, then do a print. You will notice that the nozzle moves that amount up. 

It is this bed leveling probe which is just inaccurate - though I believe we fixed that mostly with CF5 beta (_not_ the ProbeHeatersOn download).

gleonard3, 2021-02-05:
Great!  So this shouldn't concern me then?  https://i.redd.it/7q2aalmxepf61.png  I'm on a new Max with CF5.  When I got my SE over the Summer, this caused havoc for me.  Obviously Pre-CF.

Sebazzz, 2021-02-05:
No, in my opinion not 🙂

Met vriendelijke groet,
Sebastiaan Dammann

gleonard3, 2021-02-05:
Well in that case, I'm happy to adopt your opinion as my own.  Thanks!

closed 2021-02-05 by gleonard3