CR6Community/Marlin #180: Nozzle temperature lowered following start-of-print homing

coreyostrove, 2021-02-11:
### Description

I encountered a bug while printing earlier in which the following the automatic start-of-print homing I noticed that the temperature of the nozzle had been reset from 230C to 120C and the  bed temperature had been reset from 70C to 50C. This issue sounds similar to that reported in #132, and may have been addressed by the bugfix Sebastiaan mentioned there, but in that report the user explicitly stated that their indicated that their nozzle temperature hadn't changed from the set value.

### Steps to Reproduce
To the best of my recollection, the sequence of steps I performed from a fresh power-on to the bug were:

1. Perform a round of automatic bed leveling (I'd just put a fresh coat of blue painter's tape on the bed). I made no adjustments to my stored z-offset.
2. Change the value of the nozzle and bed temperatures for the "Preheat PETG" process
3. Run "Preheat PETG" and let the system come up to temperature
4. Initiated a print from the SD card. The temperatures for print were the same as those set in the preheat PETG settings.
5. The system performed auto-homing and began printing, at which point I noticed the lack of filament being extruded and saw that the temperatures had been changed.

Below is a photograph:
[image]

This happened on the v4.5.2 motherboard, using the stock creality screen and the CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19 firmware. 

I'll include this here, since it happened immediately after this bug, though it may be unrelated. Sorry that I don't have any visual documentation for this one. After noticing the bug I paused the print and attempted to manually set the nozzle temperature to the correct value. The nozzle temperature didn't respond to my manually setting it's value and stayed fixed at 120C. I then decided to cancel the print and try again. After selecting the print cancellation option and confirming that I wanted to cancel the OS locked up and had to be power-cycled to proceed.

Sebazzz, 2021-02-11:
Please retry this with CF6-Pre2 to check where the issue strands.

coreyostrove, 2021-02-11:
I've re-flashed up to the CF6-Pre2 version of the firmware (along with the firmware upgrade for the stock touchscreen), and so far haven't seen this issue pop-up, but I'll keep an eye out. Intermittent bugs are pretty insidious sometimes. I have recreated the bug involving the system locking up after trying to cancel a paused print about a half-dozen times since this post, so I'll flag a separate issue for it.

Sebazzz, 2021-02-13:
Alright - closing this. Feel free to open a new issue if the issue arises again.

closed 2021-02-13 by Sebazzz