CR6Community/Marlin #181: Firmware locks up when trying to cancel a paused print

coreyostrove, 2021-02-11:
__Issues that do not follow this issue template will be closed__

### Description

When trying to cancel a print from the touchscreen for a print that is currently paused the system locks up and needs to be power cycled to recover. 

### Steps to Reproduce

1. Pause a running print
2. Attempt to cancel the print while it is paused.
3. Select confirm on the cancellation pop-up menu.
4. At this point the printer neither cancels the print nor responds to any touchscreen inputs and needs to be power cycled.

**Expected behavior:** 

The print to be cancelled.

**Actual behavior:** 

The system locks up.

### Additional Information

Using firmware version CF6-Pre2-cr6-se-v4.5.2-mb-2021-01-27-23-23 on the v4.5.2 motherboard with a stock creality touchscreen. I'll get some photos posted tomorrow morning for this issue.

Bizzmo, 2021-02-11:
same, noticed today when doing pause.

Sebazzz, 2021-02-13:
Confirmed.

Sebazzz, 2021-02-13:
Fixed for next prerelease.

closed 2021-02-13 by Sebazzz