CR6Community/Marlin #183: Direct enter a known E-Step value on the Printer

JamesRayn, 2021-02-14:
### Description

After flashing the new CF6 Pre 3 and reseted my printer to factory settings I noticed that I can't change the E-Step value to my known good value directly on the printer without the mesurement menue. 

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Go to Control > E-Steps
2. you are able to set a new value in the "current E-Step" line, but you can't save it

Sebazzz, 2021-02-14:
Thanks for the suggestion. This was implemented previously in the e-step screen but was removed when we switched to two pages.

I've re-implemented it. Simply go to the e-steps screen, enter a value and then press "back".

closed 2021-02-14 by Sebazzz

luds66, 2021-02-20:
That would be great to have.

grobux, 2021-02-20:
@luds66 It will be part of the next release.

A little preview:
[image]

luds66, 2021-02-20:
Yep that looks nice.

Sebazzz, 2021-02-22:
There is also #189 now where you can also set the steps/mm :)

Thinkersbluff, 2021-03-09:
This issue still seems to exist within the Nightly version of 9 March 2021:
  
1. Select Esteps, 
2. tap on the current esteps value 
3. screen changes to Extruder settings menu, 
3. tap on Esteps value-> keyboard pops up, 
4. type new esteps value and tap green checkmark
5. Keyboard is removed, but current esteps value has not changed.

Sebazzz, 2021-03-09:
I can't reproduce that, but there is a slight delay. Give it some time to update.

Thinkersbluff, 2021-03-09:
> I can't reproduce that, but there is a slight delay. Give it some time to update.

Try typing in a value of 140.7 to replace the 93?

I was able to replace 140.7 with 93.0 no problem.  Can't get back to 140.7, but I can enter values up to 100.

aNOTE: I am also able to edit the Acceleration factor to replace 1000  with  100, but I cannot re-enter 1000.  Keyboard limits entries to 3 digits?

Sebazzz, 2021-03-09:
Yes, this is when you define the FloatNum to X you need to take X decimal places into account in DWIN. I didn't 😉

3DFreezeMe, 2021-03-10:
I have similar behavior when trying to manually change the E-steps. Cannot enter/save higher numbers than 102 (I need 683.17 for my Orbiter extruder).

I can type numbers up to 5 characters (incl '.' like 683.2) ok, but is will not save it. 
Upon verifying and testing I can do 102 max, which saves OK. Trying to do a 103 (or anything higher with 3 digits) does not work anymore.

When I do a M92 E683.17 from a terminal, it shows up OK _as 683.2) in the GUI.
I have this behavior on a 4.5.2 mb with both the CF6-Pre4-cr6-se-v4.5.2-mb-2021-02-28-13-49.zip and a later nightlky release. 
I did a factory default after flashing and also did flash the TFT to the correct version each time.

commit 5d3ce1c0 (2021-02-14): Allow entering e-steps directly and returning - fixes #183