CR6Community/Marlin #184: When starting print after ABL, nozzle temperature set point does not change to print temperature in gcode

DeivaSR, 2021-02-15:
### Description

Bug: Starting a print immediately after running ABL causes nozzle temperature set point to remain at ABL hot end temperature set point

### Steps to Reproduce
1. Run ABL (used default ABL temperatures for nozzle and bed, i.e. 145C and 50C)
2. Once ABL completes, start a print immediately
3. Observe the nozzle temperature set point

**Expected behavior:** 
The nozzle temperature set point should change to print temperature set point (e.g. 200C)

**Actual behavior:** 
The nozzle temperature set point remains at 145C

**Video of issue**
[Video of issue](https://youtu.be/fQA2PUNUqeM)

### Workaround
Change the nozzle temperature set point manually. Or stop the print and start it.

### Additional Information
Printer: CR-6 SE, Board v4.5.2, Stock screen
Note: Printer is using Creality stock components - no mods

Firmware version: Community Firmware Rel 6 PreRel 3
https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6-pre3/CR6Comm-Rel6-Pre3-cr6-se-v4.5.2-mb-2021-02-13-17-42.zip](https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6-pre3/CR6Comm-Rel6-Pre3-cr6-se-v4.5.2-mb-2021-02-13-17-42.zip)

Sebazzz, 2021-02-15:
Please upload a gcode file that produces the issue. Also take a picture of your settings or dump of M503. Also include some logs. Thanks.

DeivaSR, 2021-02-15:
Its been a weird few weeks for me  I decided to just re-flash the printer with the PreRel 3 f/w again, and now the problem has gone away. I am not sure what is going on with my board. Love it but hate it. Will close the issue.

By the way, just as a general feedback, if any of the devs read this, all of you are doing a great job with the f/w along with the features you are brining into it and personally, I can say this PreRel 3 is amazing!!!

closed 2021-02-15 by DeivaSR

Sebazzz, 2021-02-15:
Thanks for the feedback. v4.5.2 boards are sometimes just... weird.