CR6Community/Marlin #185: Ability to change thermistor tables for high temp hot ends.

craigrusso, 2021-02-15:
### Description

Would it be possible to be able to load new thermistor tables from the touch screen or switch between tables to support high temp hot ends like the mosquito?

Sebazzz, 2021-02-15:
Are you referring to the functionality of [`M305`](https://marlinfw.org/docs/gcode/M305.html)? That would be possible, but then you're only half of the way there. You'd still need to recompile the firmware to allow those higher temperatures, and then at the point you might change the thermistor yourself from `Configuration.h`, agree?

craigrusso, 2021-02-15:
Ah ok. I didn’t know if it where possible to make that functionality without having to recompile the firmware. Thank you though.

closed 2021-02-15 by Sebazzz