CR6Community/Marlin #186: Release 6 Pre 3 - Preheat button broken

thaukap, 2021-02-16:
In Prerelease 3 for me the prehead button (PLA) is broken. Checked it multiple times. Did a factory reset after flashing, leveled the bed, nothing else.

Steps to reproduce:

- Turn the machine one
- Press "Prepare"
- Press "Preheat (PLA)"

Expected result: Prehead nozzle and bed starts

In fact the prehead process did not start and the "Back" button did not work anymore. When you go deeper, i.e. "Movement", the "Back" button still isn't working.

Sebazzz, 2021-02-16:
Sorry, I cannot reproduce it. Make sure the touchscreen firmware is flashed correctly or collect logs otherwise.

thaukap, 2021-02-16:
The touch screen firmware is updated, I suppose otherwise I would not see the bed leveling mesh.

I will check again, with the USB port detached, as soon as the current print is ready. And including another factory reset.

Sebazzz, 2021-02-16:
Try reflashing both firmware. Rename the file name part of the motherboard firmware before doing so. 

I can't reproduce the issue so this seems like total firmware corruption to me - especially because there are no blocking actions linked to that particular button.

grobux, 2021-02-16:
@thaukap Can you check your PLA settings ?

- Control
- Temperature
- Preheat PLA settings
- Check/set, save and retry.

artworkvideo, 2021-02-16:
4.5.2 board...all the preheat functions work for me...in fact everything works ...I've tried just about everything  e-stops, PID, both ABL's.. works great...J07 printer that I received 8/26

Sebazzz, 2021-02-17:
That confirms it then 🙂

thaukap, 2021-02-17:
It's hard to reproduce, yes, but I was able to do it one more time. What I did this time:

(1) Factory reset
(2) Setup PLA heating temperature set to 220/60 and saved it
(3) Executed bed level
(4) Hit "Prepare/Preheat": No heating and the back button didn't work anymore.

When I turn on the machine and restart it the moment the screen goes black somehow something internal must be preserved. It will fail then. When I turn off the machine for a longer time, 10-20 seconds or so, it start working again.

But: That doesn't happen all the time. Even more worse. 

Right now when I start the machine from scratch it does work and the machine prints pretty nice. So I would name this a minor glitch, nothing serious and not worst to hunt it down. IMHO.

thaukap, 2021-02-17:
Therefor I close the issue.

closed 2021-02-17 by thaukap

Sebazzz, 2021-02-17:
> When I turn on the machine and restart it the moment the screen goes black somehow something internal must be preserved. It will fail then. When I turn off the machine for a longer time, 10-20 seconds or so, it start working again.

Black? Sounds rather like a glitch or other issue in your particular DWIN touchscreen hardware.

thaukap, 2021-02-17:
Fixed it. Turn off then turn on as soon as the screen goes out.