CR6Community/Marlin #187: Unable to flash printer back to Community Firmware

sjdrc, 2021-02-17:
__Issues that do not follow this issue template will be closed__

### Description

I upgraded my CR-6 SE to community firmware release 5 last week. I later flashed klipper to give it a go.

After playing with it for a while I wanted to go back to the community firmware, but somehow I managed to ignore the warning that states
> Make sure you pick the right firmware package for your particular hardware configuration!

and put the CR-6 Max firmware file on the SD card.

Now whenever trying to flash the correct firmware, nothing appears to happen, the touchscreen (which was flashed previously with community firmware) just displays "Creality" with an empty loading bar.

Have I cooked my printer?

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Ignore warning to pick right firmware package
2. Can't flash my printer any more
3. ?
4. Not Profit

**Expected behavior:** 

Printer will be reflashed after turning it on with the correct firmware file on the SD card

**Actual behavior:** 

Doesn't appear to have reflashed correctly

**Logging:**

N/A

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

Stock printer, 4.5.2 mainboard.

https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip this is the firmware I am now attempting to flash unsuccessfully.

ritchiedc, 2021-02-17:
Try changing the name of the .bin file on the sd card and make sure it is the only one on the card.

Sebazzz, 2021-02-17:
Yes, try changing the .bin file name and reformatting. There have been countless issues here in the issue tracker which describe the same thing. Sometimes the v4.5.2 just flashes the firmware for 50% or so. That motherboard is really just a POS.

sjdrc, 2021-02-18:
After trying with several SD cards and different filenames I eventually tried `FIRMWARE.BIN` and that worked. This damn board is so finicky. I've read your review of the BTT board and I'm keen to get one, but they all have to come from china with a 30-45 day lead time :/

Thanks for your help.

closed 2021-02-18 by sjdrc

Sebazzz, 2021-02-18:
Thanks for the update - glad it worked out.