CR6Community/Marlin #188: Strange home screen after flashing CR-6 Beta 5 to 4.5.3 Mobo

openscript, 2021-02-17:
### Description

I've followed Toms instruction video and also think I've read all the instructions before upgrading to the Happy New Year Edition.
Hopefully this is not a duplicate and if so, I'm sorry. I have googled for quite a while. First of all, everything seems to work, I even could level and print successfully, but the home screen looks rather strange:

[image]

If I click the buttons where they are supposed to be, navigating is no problem. The other screens look fine:

[image]

### Steps to Reproduce

0. I've downloaded the `CF5Beta-cr6-se-v4.5.3-mb-2020-12-30-14-19.zip` to my computer.
1. I've formatted a 8GB Intenso card with FAT32 4096.
2. Copied the bin file on the card and nothing else was on it.
3. Inserted it into the printer and started.
4. The loading bar took longer than usual, but it started. The place where the version was displayed stayed empty.
5. I've shutdown the printer.
6. Formatted the card again the same way and copied the whole DWIN folder onto it.
7. I removed the display and the cover. Inserted the card. Made sure it doesn't touch the case or anything conductive.
8. It took a moment or two but seemed to be successful. I've compared the output with the screenshots.
9. Then I started again, it booted and the issue appeared.

**Expected behavior:** 
No distorted home screen.

**Actual behavior:** 
Home screen is distorted.

**Logging:**
None

### Additional Information

[image]

[image]

Sebazzz, 2021-02-17:
Wow, that's pretty odd. Make sure you've started with a clean SD card when flashing the touch screen. This result is typically seen when older files reside on the SD card.

openscript, 2021-02-17:
I don't know what should be the resolution of this. It's just strange.

[image]

The SanDisk card always makes the screen faulty. Glitches or it suddenly freezes. I really wish to know why this happens, but this mystery is probably hard to uncover..

Thanks everybody for trying to help me out and also for creating the community edition :)

closed 2021-02-17 by openscript

Sebazzz, 2021-02-17:
Thank you for your update. I will update the flashing instructions.