CR6Community/Marlin #189: [FR] Allow more tuning of axis settings

Sebazzz, 2021-02-17:
Users should be allowed to adapt their axis settings.

First mock-ups:
[image]

[image]

Sebazzz, 2021-02-18:
Ready for further graphics fine tuning.

JamesSF69, 2021-02-18:
*whispers, change the red bar color!  LOL I still think it's an error when I glance over  LOL

Sebazzz, 2021-02-19:
**Movement screen**
[image]

Each axis goes to the axis screen. The axis screen dynamically changes title

**Axis screen**
[image]

The following button is only available on BTT boards and black/hidden otherwise:
[image]

**TMC settings**
[image]

**Advanced settings**
[image]

**Icons**
This is the base line now.

@grobux If you're having a go at these screens I've tried to use icons from fontawesome which you use iirc:

- Steps/mm and travel: https://fontawesome.com/icons/arrows-alt-h?style=solid
- X/Y icons: similar to that arrow
- Z: Z-offset icon
- Advanced settings and hybrid threshold: https://fontawesome.com/icons/wrench?style=solid
- Stepper current: https://fontawesome.com/icons/bolt?style=solid
- Stealthchop: https://fontawesome.com/icons/volume-up?style=solid
- Acceleration: https://fontawesome.com/icons/external-link-square-alt?style=solid
- Feedrate: existing icon
- Jerk: I drew it myself. Jerk is sudden change of speed instead of using acceleration, so I figured to draw a line graph.
- TMC settings: https://fontawesome.com/icons/microchip?style=solid
- Retract acceleration: I used the extruder icon of #115 and reversed it

No comic sans 😁

grobux, 2021-02-19:
Thanks, I'm on it !

Sebazzz, 2021-02-19:
@grobux One thing I forgot is that I added the button to access this screen next to the info button. But that is just because it is the only logical place to put it and there isn't any space elsewhere😉

Sebazzz, 2021-03-16:
@ztakis Good work on this pull request! https://github.com/CR6Community/CR-6-touchscreen/pull/41

I found a few small typo's, other than that, everything looks cool.

Sebazzz, 2021-03-16:
With this PR this is pretty much complete: https://github.com/CR6Community/CR-6-touchscreen/pull/42

There is the TMC settings button, might be a minor thing.

closed 2021-03-16 by Sebazzz