CR6Community/Marlin #190: Can anyone help with cr6se screen on ender 3pro 4.2.2 board

3dprinte, 2021-02-18:
__Issues that do not follow this issue template will be closed__

### Description

*Description of the bug or requested feature*

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** 

[What you expect to happen]

**Actual behavior:** 

[What actually happens]

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

Sebazzz, 2021-02-18:
Sorry, the CR-6 firmware is not purposely compatible with the Ender 3. I suggest you contact your seller for the correct firmware.

closed 2021-02-18 by Sebazzz

3dprinte, 2021-02-18:
this can can be done speaking to smith3d now don't need that negativity bro u can stay simple i wil always find a way to improve things