CR6Community/Marlin #191: z offset buttons will not move the gantry up or down.

bremmon75, 2021-02-18:
### Description

*Description of the bug or requested feature*
For the first 3-4 weeks the firmware worked flawlessly, now the z offset buttons do not move the gantry up or down at all. If I re install the stock firmware from creality, it works fine again. I tried to reflash with the 4.5.2 firmware again and the same issue, the gantry does not move. from the offset buttons. I can go to the move menu and move it there by .01-.10 with no trouble.
### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. [First Step] Installed the firmware
2. [Second Step]Factory reset
3. [and so on...]restart the machine
4. Level the bed and home button.
5. check the z offset is moving the printer head.

**Expected behavior:** 

z-offset to move the gantry up or down

**Actual behavior:** 

No movement with the offset buttons.. however the Move menu buttons will move he gantry.
**Logging:**

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?] 

Stock set up Serial Number P202j07K01405

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

no

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
CF5Beta-cr6-se-v4.5.2-mb-2020-12-3-14-19

Sebazzz, 2021-02-18:
Try verifying that the steppers actually do move (not visually). Touch the couplers while you press the buttons. Also make sure you've homed before you touch the buttons. Reflash the firmware and rename the bin file before doing so.

Since you mention it works fine on the same version, if it actually happens, it is not a bug but a fault in your board.

bremmon75, 2021-02-18:
The stock firmware works without issue, The steppers do not move, but I can hear a noise as if they are trying but they do not move. As said in the post I did home first. I will rename and re-flash to see if that clear it up.

bremmon75, 2021-02-18:
changed the bin name and re-flashed, same issue still. I can raise and lower them in the Move Menu by .01 but the z offset buttons do nothing.

Sebazzz, 2021-02-18:
I can't do anything, this is hardware related. Note that stock moves at 0.05mm so that is a _lot_ more steps. As you noted, the same binary firmware worked fine before.

Sebazzz, 2021-02-18:
One thing you can try is trying the 6 prerelease 3 in the releases section and check if that does anything.

ritchiedc, 2021-02-18:
For a point of reference, I have mb 4.5.2 running 6 prerelease 3 and the z moves for me using the leveling menu. I turned on the printer, selected leveling and then moved the z up and down without issue.

bremmon75, 2021-02-19:
Same thing on the Community Firmware release 6 prerelease 3 (It's freakin amazing). If I use the move menu I can move the print head up and down by .1mm. But when I try to move the z-offset the head does not move. I guess I'm stuck with the stock firmware then?  I'm sad, you guys have done such a great job with this community firmware. Thank you for your time and assistance.

closed 2021-02-19 by Sebazzz