CR6Community/Marlin #193: [FR] Implement settings for filament runout and PLR

Sebazzz, 2021-02-19:
### Description

Allow turning filament runout sensor on/off and PLR.

Sebazzz, 2021-02-19:
I can't find an icon, so using this:

- Filament runout: https://fontawesome.com/icons/cut?style=solid
- Power loss recovery: https://fontawesome.com/icons/poo-storm?style=solid

Sebazzz, 2021-02-19:
Suggested icon for PLR by a Discord user:

[image]

Sebazzz, 2021-02-19:
This is implemented too. @grobux I had no place to put a button for this screen so I've now put a hidden button on the title bar of the control screen.

ztakis, 2021-02-22:
As a temporary (or not) solution, you might want to use the Creality logo at the home screen as a button that leads to the info screen and then change the info button in the Control screen to Misc. settings

Sebazzz, 2021-02-23:
@grobux I've decreased the size of the "reset to factory settings" button, and moved the info button to the left of it.
That makes room on the control screen to add a "misc settings" button. I'm actually pretty satisfied with this solution. Needs some final alignment checks of course, and proper fonts, but I'Il leave that to you 💯

Sebazzz, 2021-03-16:
Thanks to this PR by @ztakis this has now been fully implemented: https://github.com/CR6Community/CR-6-touchscreen/pull/40

closed 2021-03-16 by Sebazzz