CR6Community/Marlin #195: [FR] After ABL has completed, stay on mesh display screen until Back is selected

ztakis, 2021-02-22:
### Description

After ABL has completed, stay on mesh display screen until Back is selected.
Also as a confirmation that ABL has run and completed successfully.

JamesSF69, 2021-02-22:
> ### Description
> After ABL has completed, stay on mesh display screen until Back is selected.
> Also as a confirmation that ABL has run and completed successfully.

Yep, I think this might be nice. We do have a "view mesh", but maybe just a delay instead of an immediate change of screen at the last probe.

Sebazzz, 2021-02-22:
Agree, now we have the view mesh screen this screen has actually some value so we might as well remain on the screen.

Will still return immediately after the last point when leveling from a print though.

closed 2021-02-22 by Sebazzz

ztakis, 2021-02-22:
If you agree then can you please also disable steppers after ABL completes? Thank you !

commit e1752c40 (2021-02-22): Stay on the leveling screen after leveling when in the leveling workflow - fixes #195