CR6Community/Marlin #197: Implement fade height settings on the DGUS screen

Ranney1, 2021-02-24:
Is it possible to implement these settings on the DGUS screen? Maybe on screen 72?

#if ANY(MESH_BED_LEVELING, AUTO_BED_LEVELING_BILINEAR, AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  **#define ENABLE_LEVELING_FADE_HEIGHT**
  #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
    #define **DEFAULT_LEVELING_FADE_HEIGHT 10.0** // (mm) Default fade height.
  #endif

Would be great if it is possible to turn it on/off and set the default leveling fade height!

Sebazzz, 2021-02-24:
*Klaar terwijl u wacht*

@grobux I added an extra control in the leveling settings screen. The font probably needs some improvements, otherwise it is ok (and of course your source PSD needs to be updated as well).

Sebazzz, 2021-03-16:
Thanks to this PR by @ztakis this has now been fully implemented: https://github.com/CR6Community/CR-6-touchscreen/pull/41

closed 2021-03-16 by Sebazzz