CR6Community/Marlin #198: Request feature?

Daskew78, 2021-02-24:
I would like to request that the "Extrude" feature has a speed setting added and the set speed is saved automatically for future uses.

I'm told following issues I've been facing that the current "Extrude" feature is mainly for getting the filament from the extruder to the hotend which works fine, but I (and others no doubt) need to simply extrude some filament at times.

The issues I faced using the "Extrude" feature was that the extruder was clicking and missing because it can't push the filament through the hotend quick enough. I even changed my extruder to the official Creality MK8 but the issue still remains.

If I can set the speed and slow it down it would resolve the issue and allow me (and others) to simply extrude a set amount at a suitable speed.

Thanks for your consideration.

astralaster, 2021-05-27:
I can confirm the clicking while extruding with the stock extruder. But it does not occure everytime I use the load filament feature. Im using release 6.

Z3R0-89, 2021-07-04:
I am also having issues with the extruder clicking while loading filament. Does exist a command to change that speed via a serial terminal?

GKai1, 2021-07-05:
A related problem: Loading filament works fine for me, but unloading doesn't. Unloading happens at a much faster speed than loading currently. Combined with the gear ratio of a BMG extruder, this makes the gears spin so fast that they grind through the filament when unloading.

My workaround is unloading bits of 6 cm of filament several times, so the extruder does not have time to pick up full speed. But this is annoying, so I very much hope for an opportunity to customize unloading speed in V7.

schattenphoenix, 2021-07-28:
I am having a similar issue, using the Orbiter extruder the moter just cant keep up and misses all steps over around 32mm/s when unloading (which accelerates quick).

I found a temporary fix by setting the maximum extrusion speed in the movement settings but found another problem that the setting seems not to be stored for me.

Will add another issue regarding the not stored maximum feed rate.

BloodyRain2k, 2021-08-05:
On the topic of filament un/loading: I would really like to see an option to skip the additional movement shenanigans that the CR-6.1 does:
- moving the Z up
- doing whatever for 5 seconds (it's not waiting for the heating, I checked that and it was already at temperature)
- THEN finally un/loading the amount of filament I've set
- then more shenanigans
- moving Z down again
and only then it's finally done.

Often I just want the filament moved by a few mm to either stop the drooping early or manually prime it lightly, and being locked out of proceeding while waiting on these additional, and in my opinion at that moment useless, movements is just frustrating.

I understand that the priming and jerking it does for unloading is so there's as little as possible stuck back in the heating zone.
But yeah, if I'm only retracting a minimal amount because of drooping, it's a complete waste, especially given that unloading forces a priming in the first place for that.

SANGERA2, 2021-12-18:
> On the topic of filament un/loading: I would really like to see an option to skip the additional movement shenanigans that the CR-6.1 does:
>
> * moving the Z up
> * doing whatever for 5 seconds (it's [...]

Definitely +1 to this. Personally I think it should be off by defaul too.

I was surprised after moving from the standard CR6 firmware that there is no option to feed/retract filament without time wasted moving the hotend around. If I want to push 10mm of filament through at a time until I see the colour change complete it takes AGES now!

Thinkersbluff, 2021-12-18:
The Marlin LCD menu provides "Move Extruder" for the type of movement missing in CF. 
The Load/Unload filament feature is implemented using the Nozzle Park functionality, which enables the option of changing filament mid-print.
All programming and maintenance of the CF is on a purely volunteer basis, with a published open invitation to anyone able and willing to address these open issues.
Alternatively, if you use Octoprint, you may find you can do that sort of movement using the Octoprint Control screen.

jc265, 2022-09-13:
So is there no way to slow down how fast the change filament function is?  I did my first print tonight that had a filament change mid print.  After loading the new filament, my extruder sounded like a machinegun, constantly skipping when trying to feed in new filament.  I'd like to figure out a way to make that stop for future prints with filament change mid print.

Thinkersbluff, 2022-09-13:
No, the CF6.1 interface does not include a "move extruder" menu.

The change filament feature uses Marlin code M600, which is documented here: 

https://marlinfw.org/docs/gcode/M600.html

There is no speed parameter passed with M600 at runtime.  
The programmed behaviours of M600 are predefined at compile time, in the Advanced Pause section of Configuration_adv.h, documented here:

https://marlinfw.org/docs/configuration/configuration.html

The stock hotend cannot melt filament fast enough to extrude at the default feed rates programmed into M600. Load/Unload (M701/M702) use those same rates, with the same issues.

If you are able to compile the firmware, you can experiment with changing those settings to work better with your configuration.

Alternatively, you can release the stock extruder lock and manually push or pull the filament, to purge the old colour before resuming the print.