CR6Community/Marlin #199: Move screen's back button doesn't work when 1mm steps are selected.

ztakis, 2021-02-26:
### Description

Move screen's back button doesn't work when 1mm steps are selected.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Navigate to the Move screen.
2. Select 1mm steps.
3. Press the back button.

**Expected behavior:** 

To return to the Prepare screen.

**Actual behavior:** 

Stays at the Move screen

**Logging:**

Send: M111 S247
Recv: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
Recv: ok
[...]
Recv: Invoking handler for screen 28with VP=4098 value=2
Recv: SetNewScreen: 42
Recv: SetNewScreen (saving): 42
[...]
Recv: Invoking handler for screen 42with VP=4158 value=3
Recv: SetNewScreen: 43
Recv: SetNewScreen (saving): 43
[...]
Recv: Invoking handler for screen 43with VP=4166 value=2
Recv: SetNewScreen: 44
[...]
Recv: Invoking handler for screen 44with VP=4182 value=3
[...]
Recv: Invoking handler for screen 44with VP=4182 value=3
[...]
Recv: wait
[...]
Recv: Invoking handler for screen 44with VP=4182 value=3
[...]
Recv: Invoking handler for screen 44with VP=4166 value=1
Recv: SetNewScreen: 43
[...]
Recv: SetNewScreen: 42

### Additional Information

CR6Comm-Rel6-Pre4, mb 4.5.2

Ranney1, 2021-02-26:
@Sebazzz The VpAddr and button value on this screen is different.

@ztakis if you want, you can fix this in the dgus tool by chaning the button value on this screen to '0000' and the VpAddr to '2295'

ztakis, 2021-02-26:
@Ranney1 I am not comfortable with the dgus tool yet but thanks. I now understand what is wrong.

Sebazzz, 2021-02-26:
Thanks, got it! 

@ztakis I wasn't sure whether you wanted to have a go at the DWIN touch screen as well on top of graphics/PSDs. If you don't, well, this issue is fixed now and once you have the graphics ready I'Il do the DWIN part. Otherwise you can either do what @Ranney1 says (if you have already started) or pull the latest sources from the touch screen repository.

closed 2021-02-26 by Sebazzz