CR6Community/Marlin #200: Nozzle Temp Changes on z direction change

stretch327, 2021-02-26:
### Description

In the bed leveling function, the bed and nozzle are heated to the appropriate temps, the head moves to the first location and then as the nozzle moves downward, the LCD says that the target temp has been set to 0°. Once it touches and starts moving back up, the target nozzle temp now reads 120°. For each position in the leveling, it seems to lose one or two degrees because it never really gets back up to 120° before moving on to the next position.

### Steps to Reproduce

Do a bed leveling and watch the nozzle temp on the LCD screen

**Expected behavior:** 

I expect the nozzle temp to stay the same throughout the process so it isn't fluctuating.

**Actual behavior:** 

The temp fluctuates between 120° and 110° with a slight downward trend.
**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

CF5Beta-cr6-se-v4.5.2-mb-2020-12-30-14-19.zip

https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.3-cr6-community-release-5-beta

Sebazzz, 2021-02-26:
Thank you for your report.

This is expected and will not be fixed because the accuracy is not affected by the nozzle getting a few degrees colder. Otherwise you'd need to wait for the nozzle to get back to temperature which would significantly increase the time needed for leveling (up to 30 minutes I think).

closed 2021-02-26 by Sebazzz

stretch327, 2021-02-27:
I don't see how that's possible. The creality firmware maintains the temp and it actually took _less_ time than this does. 

That said, I found today that the nozzle temp is also set to zero when a print is started, just after the head centers, long enough that 20 degrees are lost and it takes another 3 minutes to get that heat back.

ritchiedc, 2021-02-27:
I haven't looked at the implementation but from the way it works I suspect that the z homing monopolizes the cpu to improve accuracy and repeatability. I prefer this over saving a bit of time. I have learned to make sure the nozzle is close to the bed to reduce the temperature deviation.

Sebazzz, 2021-02-27:
I think I misunderstood you. I assumed that you were aware of these sections in the release notes:

> Increase bed leveling and homing (Z-offset) accuracy
>    We do this by preheating before probing and turning off heaters when probing so electrical interference when reading the strain gauge is minimized
>   Homing and probing is perceived slower - but in terms of print time this is just a fraction of the time
>    By default we probe at 120C/50C - but if you have preheated your printer to a higher temperature, then that temperature is kept
>    Disable this behavior by commenting PREHEAT_BEFORE_PROBING and PROBING_HEATERS_OFF in the firmware configuration

And

> As mentioned above, while touching the bed we turn off the hot-end momentarily. If you don't want that, you can use the builds below. Only use these if you have a good reason to do so.
> [file listing here]
> Note: The files with ProbeHeaterOn keeps the hot-end heated while probing - in our experience this decreases bed leveling accuracy and cause a drifing Z-offset - but some people prefer it this way. You have a choice.

In CF6 this will be behind a preference (via #106). Note the numbers are significantly more inaccurate when probing with the heaters on:

**Mesh with heaters on**
```
Recv:       0      1      2      3
Recv:  0 -0.009 +0.009 +0.060 +0.171
Recv:  1 +0.003 -0.072 +0.011 +0.136
Recv:  2 +0.165 +0.027 +0.147 +0.295
Recv:  3 +0.499 +0.356 +0.405 +0.628
```

**Mesh with heaters off**
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 +0.139 +0.107 +0.115 +0.222
Recv:  1 +0.127 -0.032 +0.021 +0.135
Recv:  2 +0.226 +0.055 +0.128 +0.255
Recv:  3 +0.510 +0.312 +0.337 +0.544
```

Quite a significant difference! The point at (0,0) is almost a layer height different!