CR6Community/Marlin #203: When canceling the mesh validation the print head touches the already printed patches

KoenVanduffel, 2021-02-28:
When pressing cancel the printer stops the currently printing patch and moves to the next (that was a little unexpected but nice behavior). When canceling the last patch the head moves back to 0,0 quickly without increasing the z-height first and thus touches the already laid down patches (they were squished a little too much so were sticking just above the print head position.
It was too little to damage anything but still touched the already laid down PLA.

Sebazzz, 2021-03-04:
Well, I tried everything but I cannot convince it to raise Z first and then move X and Y. On hold for now.

closed 2021-03-05 by Sebazzz

KoenVanduffel, 2021-03-07:
Thanks, understandable this is not a priority thing if its not straightforward.