CR6Community/Marlin #204: Auto Bed Levelling settings not restored after G28 Homing

etaardvark, 2021-02-28:
__Issues that do not follow this issue template will be closed__

### Description

Auto Bed Levelling settings not restored after G28 Homing

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Start Print with G28
2. After G28 Bed Levelling appears to be disabled
3. Print fails due to failed first layer.

Also manually ran the following Gcode to test

G28 ; Home
G29 ; AutoLevel - Noted that the bed levelling was saved
M420; returns that Bed Levelling is ON
G28; Home again
M420; Returns that Bed Levelling is OFF

**Expected behavior:** 

I *thought*, though happy to be corrected, that with the community firmware the bed levelling was enabled after G28

Can add M420 L1 S1 to force it to be enabled in my startup G-Code.

Running M420 after a restart shows the bed levelling disabled.

**Actual behavior:** 

Bed Levelling is disabled after G28;

**Logging:**

### Additional Information

Community Firmware V5
Board Version 4.5.2

Sebazzz, 2021-02-28:
Please try it on the latest prerelease to see where the issue stands. 

However after boot up the mesh always disabled.

Sebazzz, 2021-03-04:
Feel free to reopen if the issue still happens.

closed 2021-03-04 by Sebazzz

NunchakusLei, 2021-04-08:
Do you know if the mesh will be stored in the internal memory or the SD card?