CR6Community/Marlin #205: M600 extrudes a blob after priming

ThisIsMyUniqueUsername, 2021-03-01:
### Description

Using M600 for filament change mid-print often extrudes a blob before the layer starts. The command first retracts the filament, then after loading a new one extrudes some to prime the nozzle and get rid of the previous colour. After priming it retracts a bit, moves to the print to start next layer, and then most times it extrudes a blob.

I tested it with different models.

To work around it I always add a small sacrificial cube that prints before the main model, so the blob goes on it.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Slice model with M600 
2. Change the filament as instructed, continue print
3. Wait for the printer to prime the nozzle
4. Watch the head move to start a new layer and extrudes a blob before continuing

**Expected behavior:** 

I would expect the next layer after filament change to print without the blob

**Actual behavior:** 

It appears that a bit too much filament is extruded before new colour layer starts.

**Logging:**

### Additional Information

My Gcode:

G0 X104.434 Y63.373
G0 X104.434 Y62.294
G0 X103.848 Y61.708
;TIME_ELAPSED:11209.707360
M600 E30.00 L100.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
;LAYER:22
M117 INDICATOR-Layer22
;TYPE:WALL-INNER

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[image]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

Release 5

Send: M115
Recv: FIRMWARE_NAME:Marlin CR-6 community firmware (Dec 30 2020 14:19:16) SOURCE_CODE_URL:https://github.com/CR6Community/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:CR-6 Community EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff

[image]

Sebazzz, 2021-03-01:
This is not extruding (the E-axis doesn't move), this is pressure in the nozzle causing extrusion. You need to fix that in your work process, can't fix this in firmware.

Sebazzz, 2021-03-01:
Since this is native Marlin functionality I suggest you open an issue upstream.

closed 2021-03-01 by Sebazzz

ThisIsMyUniqueUsername, 2021-03-01:
The extruder moves right after the head is positioned over the model, it extrudes some amount of filament and then starts moving the head. I assumed it's just a behaviour of the M600 command.