CR6Community/Marlin #206: Random Print stoppage then  failure to print ( Crashes / restarts at nozzle wipe )

samblissett, 2021-03-01:
__Issues that do not follow this issue template will be closed__

### Description

Print randomly stops as if paused untill power is cycled then asks to resume. If resumed it does not heat and tries to home.

After this happens the next prints go through the warm up process and moves to bottom left corner to perform nozzle wipe but instead stops and the loading bar appears on screen as if it has just restarted and goes back to standby.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Print and await random stoppage
2. Clear and start new print
3. Observe software crash

**Expected behavior:** 

Print object selected

**Actual behavior:** 

Printer seems to crash.

**Logging:**

Unfortunately the printer is located in the garage and i do not have capacity to run pronterface on a laptop or octoprint.

### Additional Information

Stock 4.5.2 board had this issue and would not resolve after multiple firmware flashes. I upgraded to BTT CR6Se board and tried release 5 which worked for a few prints then the same issues occurred. Then tried the new Beta release 6 build ( yesterday ) and got a partial print before the same issue occurred.

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6-pre4/CF6-Pre4-btt-skr-cr6-with-stock-creality-tft-2021-02-28-13-48.zip

Sebazzz, 2021-03-02:
Thank you for your report. What makes you think this is firmware related as opposed to hardware related? 

Provide a minimal gcode sample on which the issue occurs please.

samblissett, 2021-03-02:
Running test print using Prusa slicer as opposed to Cura. These sliced files have printed flawlessly several times previously and now wont print, however as the Bed mesh validation prints without issue it would seem there may be something in the gcode that it does not like.
[SampleCode.txt](https://github.com/CR6Community/Marlin/files/6071793/SampleCode.txt)

Sebazzz, 2021-03-02:
Is this the most _minimal_ gcode sample you can provide?

Sebazzz, 2021-03-02:
@samblissett Since you appear to be running the prerelease version anyway - can you try and flash the `CF6-Pre4-cr6-se-v4.5.2-mb-no-watchdog-2021-02-28-13-50.zip` release and check how that goes?

I recommend flashing the motherboard firmware to an earlier release first (verify in the info menu that it has flashed), then flashing `CF6-Pre4-cr6-se-v4.5.2-mb-no-watchdog-2021-02-28-13-50.zip` (verify in the info menu it has flashed). 

**Caution** the "no-watchdog" version will _not_ reset the printer if the firmware freezes, and can therefore cause thermal runaway.

Sebazzz, 2021-03-11:
@samblissett Are you able to try the "no-watchodig" version?

samblissett, 2021-03-12:
@Sebazzz Sorry for delayed reply. Even though the same files sliced in Cura printed previously they will not now, and slicing again also fails half way or in most cases before the nozzle wipe. I sliced the same files in PrusaSlicer and it just works..... i am not sure what is the issue with the Cura sliced files, but PrusaSlicer is working fine, although with some files there are extended pauses between layer changes and seemingly before retraction occurs leading to over extrusions , blobs and stringing.

I don't really want to try the NoWatchDog version as the printer is in my garage and not frequently monitored.

samblissett, 2021-03-12:
I have been running the BTT SKT version 1.0 board as i believed this would have resolved my issues, but the same was true of the BTT board

Sebazzz, 2021-03-16:
Assuming resolved then 👍

closed 2021-03-16 by Sebazzz