CR6Community/Marlin #209: [FR] Allow manual bed leveling

Sebazzz, 2021-03-04:
### Description

Given some users are _still_ experiencing bed leveling issues, both on stock and CF, we should give them the opportunity to manually level their bed by overriding the mesh leveling values.

For now implement for the CR-6 SE with a 4 by 4 grid only.

Sebazzz, 2021-03-04:
This works for the most part now.

Two issues remain:
- Unable to enter negative numbers
- DWIN seems to be truncating floats (so the very last decimal might be different) *very minor*

Sebazzz, 2021-03-05:
Graphics are done by @ztakis - negative input implemented. That makes this complete.

Manual bed leveling, on the CR-6:
[image]

closed 2021-03-05 by Sebazzz

Eliasezequiel81, 2022-01-22:
hello my name is ezequiel and i'm from argentina, i bought a cr6 max and installed the community firmware and the firmware is excellent but i would like to know if someone can explain to me how the mesh works and how it is edited and how i do it to get the values that my bed needs. Thank you very much for everything you do

Thinkersbluff, 2022-01-23:
Sorry, this issue is closed.  If you are looking for help and to join the Community, please join us in the Discord, here: https://discord.gg/U5rgMCT5