CR6Community/Marlin #210: [FR] Linear advance calibration test

Daskew78, 2021-03-04:
I was wondering since we can already adjust retraction, esteps and other nice features if it's possible to add a Linear Advance test from the menu? It would need the initial test and the the ability to refine the test then store it.

I hope it's possible, just another cool feature for the already ultimate firmware.

Thanks for your consideration.

Sebazzz, 2021-03-05:
It would be nice, but I would need to study the gcode by the linear advance pattern generator in detail to check if it is feasible. There are also a lot of parameters.

Daskew78, 2021-03-05:
Thanks Sebazz, no rush, just if and when you get time it would be lovely and a brilliant addition to the firmware :)

p.s. Here's the pattern generator, I don't know if you can copy it in any way?
https://marlinfw.org/tools/lin_advance/k-factor.html

BloodyRain2k, 2023-03-11:
I'm not a fan of digging up old posts, but as this feels like the best place to ask: what version of Linear Advance is the 6.1 release using? 1.0 or 1.5?
I've been trying to find that out somehow, but I can't even find out which version of Marlin it's based on...
Given that it's quite a while since the release I would assume 1.0, but I don't know.

ritchiedc, 2023-03-11:
The appropriate place for this question is on the community discord. The info you are looking for is in version.h as follows:

#define DETAILED_BUILD_VERSION "CR-6 community firmware - release 6.1 (based on Marlin 2.0.8.1)"

The specific version of LA is what was in that Marlin version