CR6Community/Marlin #211: Z-offset display doesn't show numbers below -0.01

ztakis, 2021-03-05:
### Description

Z-offset display box doesn't show numbers below -0.01

### Steps to Reproduce

The following assumes you are using the original CR-6 bed.

1. Go to the leveling screen.
2. Place a credit card (or similar) at the center of the bed.
3. 'Home' over the credit card and then remove it.
5. Now try to adjust Z-offset using the "paper method".
6. If you are not hitting below zero, try again the above steps with a thicker card.
7. In my case, I had to press minus about 15 times below zero to start to feel the paper drag.
8. While nozzle does move lower, the z-offset display stops at -0.01

**Expected behavior:** 

It should be able to display +/- 100

**Logging:**

Recv:  T:25.00 /0.00 B:23.75 /0.00 @:0 B@:0
Send: M111 S247
Recv: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
Recv: ok
Recv:  T:25.00 /0.00 B:23.75 /0.00 @:0 B@:0
Recv:  T:25.03 /0.00 B:23.79 /0.00 @:0 B@:0
Recv:  T:25.00 /0.00 B:23.79 /0.00 @:0 B@:0
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv:  T:24.96 /0.00 B:23.75 /0.00 @:0 B@:0
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (767 bytes; crc 16099)
Recv: //action:notification Settings Stored
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (767 bytes; crc 13948)
Recv: //action:notification Settings Stored
Recv:  T:25.00 /0.00 B:23.75 /0.00 @:0 B@:0
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (767 bytes; crc 47146)
Recv: //action:notification Settings Stored
Recv:  T:24.96 /0.00 B:23.75 /0.00 @:0 B@:0
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv:  T:24.96 /0.00 B:23.79 /0.00 @:0 B@:0
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv:  T:25.00 /0.00 B:23.83 /0.00 @:0 B@:0
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (767 bytes; crc 26661)
Recv: //action:notification Settings Stored
Recv:  T:24.96 /0.00 B:23.75 /0.00 @:0 B@:0
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv: Invoking handler for screen 52with VP=4164 value=3
Recv:  T:25.00 /0.00 B:23.75 /0.00 @:0 B@:0
Recv: Saving DWIN LCD setting from EEPROM
Recv: echo:Settings Stored (767 bytes; crc 28220)
Recv: //action:notification Settings Stored
Recv:  T:25.06 /0.00 B:23.75 /0.00 @:0 B@:0
Recv:  T:25.00 /0.00 B:23.75 /0.00 @:0 B@:0
Recv:  T:25.00 /0.00 B:23.75 /0.00 @:0 B@:0
Recv: wait

### Additional Information

CR6Comm-Rel6Pre4 / v4.5.2

Thinkersbluff, 2021-03-05:
Why would you want to do this?

Z Offset is meant to compensate for a bias in Z=0.0 representing the amount of deflection required to trigger detection of the bed by the strain gauge.

If the nozzle is striking an obstacle and not the bed itself, why ask the firmware to compensate? Why not remove the obstacle?

To dial-in negative numbers on Z with Marlin, one usually needs to release the software endstops.
That option is not currently present on our menus.

ztakis, 2021-03-05:
Personally I don't, but I have seen posts in reddit and other forums, that some users (usually with some customized bed) need to get lower than zero

ztakis, 2021-03-05:
Short answer: for testing

Thinkersbluff, 2021-03-05:
I guess if we want to define limits for our support of non-stock printer configurations, this is as good a way as any to expose and address that.  I did see @Sebazzz recently exploring the idea of converting the CR-6 to a BLTouch ABL system, which would require the negative offset and release software endstop functionality.

ztakis, 2021-03-05:
If you haven't noticed, the nozzle DOES move below zero, it's the display that doesn't follow.

Thinkersbluff, 2021-03-05:
> If you haven't noticed, the nozzle DOES move below zero, it's the display that doesn't follow

I could understand citing that fact as an issue.
I see two possible solutions to that issue.  

As currently phrased, I think this "issue" is actually a feature-request, assuming a particular resolution & the _actual issue is not clearly stated.

Sebazzz, 2021-03-05:
Confirmed. Seems to be something in interop.

Sebazzz, 2021-03-05:
We seem to be transmitting the right information:

```
Z-offset: -0.50 - -50]
```

For `M851 Z-0.5`.

It must be something inside the lower interop layer. 

I will compare it with the code from CF3:

https://github.com/CR6Community/Marlin/blob/90066ad2f95dab75d64185ca5140da6ac16514e5/Marlin/src/lcd/dwin/LCD_RTS.cpp#L534-L567

Sebazzz, 2021-03-05:
This has now been fixed. Thanks for reporting.

closed 2021-03-05 by Sebazzz