CR6Community/Marlin #212: Crash/Restart on Octoprint Connect attempt

gleonard3, 2021-03-07:
### Steps to Reproduce

1. Turn printer on.
2. Attempt to connect via Octoprint
3. Observe Restart on CR-6 screen, progress bar stalls.

**Expected behavior:** 

1. Octoprint just connects as normal

**Actual behavior:** 

1. Eventually does connect.
2. Octoprint log details issue during handshake attempt

**Logging:**

2021-03-06 20:12:33,171 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2021-03-06 20:12:33,204 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 7 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyUSB0@250000, /dev/ttyUSB0@230400, /dev/ttyUSB0@57600, /dev/ttyUSB0@38400, /dev/ttyUSB0@19200, /dev/ttyUSB0@9600
2021-03-06 20:12:33,204 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 115200
2021-03-06 20:12:33,205 - octoprint.util.comm - INFO - Connecting to port /dev/ttyUSB0, baudrate 115200
2021-03-06 20:12:33,214 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2021-03-06 20:12:33,218 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:33,271 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2021-03-06 20:12:33,275 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:33,277 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2021-03-06 20:12:33,282 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:35,407 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 250000
2021-03-06 20:12:35,415 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2021-03-06 20:12:35,422 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:37,426 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2021-03-06 20:12:37,440 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:39,442 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2021-03-06 20:12:39,454 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:41,457 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 230400
2021-03-06 20:12:41,459 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2021-03-06 20:12:41,471 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:45,123 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2021-03-06 20:12:45,135 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:47,283 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2021-03-06 20:12:47,296 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:49,299 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyUSB0, baudrate 57600
2021-03-06 20:12:49,301 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2021-03-06 20:12:49,319 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:51,324 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2021-03-06 20:12:51,334 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:53,338 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
2021-03-06 20:12:53,349 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-03-06 20:12:53,809 - octoprint.util.comm - WARNING - Received line:
2021-03-06 20:12:53,810 - octoprint.util.comm - WARNING - | ....................................................................................................^^^F~^F<86>ø^F~<98><80>\x00f<9e>~fx^Xff\x00<86>^Fæ<98>^F<86>ø^F<9e><86><9e>ø<9e>\x00<98><80><86><98><86>^^^^æ<9e>æ<9e>æ<9e>æ<86>^Xf<86>^F<9e>x<98><9e>^Xføfx`^X^F<98><80>fæà^F<9e>þ<9e><98>^X^X<98>^Fæø^F~\x00<98>f^F^Fx^F^F~f^Fx<98>^X^Xx\x00^Xx^F<98><9e>^X^Xxæ<86>^X^X^X^X^F<98>\x00f~`^F<9e>þ<9e>^X<86>~\x00^X<86>^^f<86><86>àfæø^F~<86><9e>^Ff<86><9e>\x00^X`^Ff<9e>æ<9e>^F<86><9e>ø<9e>à<86>\x00<98><80><86><98>f<86>x^X<98><98>f`\x00<98>~<86><9e>`^F<9e>\x00<98>ø^F~þ<9e>^^ææ^F<98><9e>æ<9e>`^F<9e>f<86>ø<86>àæ<9e>^F<86>ø^Fx<86><86><98><80>fæà^F<9e>þ<9e><98><98><80>æ<9e>æ<9e>\x00f<9e>à<9e>f^Fx<98>^X^X`f<86>à^F^Xx<80>^X^Xx\x00^Xx^F^X^Ffæà^F<9e>þ<9e><98>^X^X`^Xfxf^F<98>^^fxæ<9e>þ<9e>^Xf<86>~\x00^X^^\x00^X^X<98>^^\x00^X^Xf<9e>^F<86><9e>ø<9e>f<86>ø<86><80>f~x<86>xf<86>ø<86><80>f^F~fæø<86>~\x00<98>à<98>^^^X^X^X^Fþ<86>þ<86>^Fæà^F~<86><9e>þ<9e>ø<9e><98>^X<9e>þ<9e>`f<9e>xf<9e>^^f^F~<86><9e>þ<9e>ø<9e>\x00<98><80><86><98><86>^^^^æ<9e>æ<9e>æ<9e>æ<86>^Xf<86>^F<9e>x<98><9e>^Xføfx`^X^X<98>fx^F^Fx<86><86><86><98><80>`f<98>~f<98>\x00f<98>`f<98>^F^X<98>þ<9e>^F^Fx<86><9e>ø<9e>~<86>^F`f<98>~f<98>\x00f<98>`f<98>^F^X<98>þ<9e>^F^Fxf^Fx<98><80>`f<98>~f<98>\x00f<98>`f<98>^F^X<98>þ<9e>^F^Fx<86><9e>ø<9e>~<86>^F`f<98>~f<98>\x00f<98>`f<98>^F^X<98>þ<9e>^F^Fxf^Fx<98><80>à<98>þ<9e>^F^Fx<86><9e>ø<9e>~^F^X`~f^X<86><98>\x00^X<98>^^^F`\x00<98>øfx`^F~<86><9e>ø<9e>~^F^Xx^Xæ<9e>æ<9e>\x00<98>^Ff`\x00f<98>æ<98>æ<98><98><80>fæà^F<9e>þ<9e><98>^Xf<80><98>x\x00<98>ø^F~þ<9e>^Xfx`^F<98>øfx`^F~<86><9e>ø<9e>~æø^F^Xøfx`<86>øf<9e>f<86>~f^Fx\x00^X<86>~^X^^~^X^Xàf^F~fæøæ~\x00<98>à<86>øæà^F<98>x~<98>x~^Xx<86><86><98><80>`f<80><86>øf<9e>ø<9e>~^F^X<86>øæ<9e>^Xfx<98><80>þ<86>þ<86>^Fæà^F~<86><9e>þ<9e>ø<9e><98>^X<86>øæ<9e>æ<9e>\x00^F~þf<86><86><9e>`<86>^F^X^F<98>^X^X^^^^^Fff<98><80>^F<9e>fæàææ^F<98>øf~^^æàfxf^Fxf^Fx^F^X^Fæ<98>^F<86>ø^F<9e><86><9e>ø<9e>àfø<86>øæà^Ffx^^æ<9e>`fxàfàf~ff~f<86><86>^^^F^F^F^^`^X^X~\x00<98><98>^Ff<98><80><86>~^Ff<9e>à<9e>^F<86>à^F<9e>f^F^Xàf~`^F^Xøfx^F^Fx\x00<98><9e><98>à^X^F<98><80><98><80>^^<98>`\x00<98><98>æ<9e>x^F~~f<86>øfx\x00^X`^Ff~à<9e>`^F^Xxf^F~fæà^F~f^Fx\x00<98>`^^^X^F^^ff~^^fx<98>^X^X<98>^F<86>ø^Fx<98><80>^Xf^X^X^X^X~\x00^X^X^F^X^X^X^X^X^X^X^X^F^Xfà^X^X^X^X~\x00^X^X<86>x\x00^X^X^X`x<86>^^x<86>^F^Xfx\x00^X^X^X~\x00^X^X^F^X^X^X^X^X<98>à^X^^<98><80>^Xfx\x00^X^X^X~\x00^X^X^F^X^X^X^X^X^X^X^X^F^Xfà^Xx\x00^X^X~\x00^X^X^F^X^X^X^X^X^X<98>\x00<86>^Fà<98>^Xf^X^X^X~\x00^X^X^F^X^X^X<98>^^f^X^^f^X^F\x00f<80>^F^X^X^X~\x00^X^X^F^X^X<98>à^X^^x^F<98><80><86>^F\x00f<98><86><98>^F^X^X~\x00^X^X<86>^^^F^X^X^X^X^X^X^X^F^^<86>`^^f<98>^F^X~\x00^X^X^F^X^X^X^X^X^X^X^X^F<80><98>xæ<98><86><98>^F^X~\x00^X^X^F^^\x00^X^X^X^X^X^X^X^F`<86>`^^f<98>^F^X~\x00^X^X^F^X^X^X^X^X^X^X^X^F^F<86>`^^f<98>^F^X~\x00^X^X^F^X^X^X^X^X^X^X^X^Fæ<98>æ<98>^F<86><98>^F^X~\x00^X^Xf`\x00^X^X<98>^F^F`^^^X^^<98><80>^X<86>`^F<86><98>^F^X~\x00^X^Xf`\x00^X^X<98>^F^F`^^^X^X^Ff^Fæà^F<98>^Xf~\x00^X^X<86>`x<86>`x<86>`x<86>`<86>^X^Ff^Fæàæ<98>^Ff~\x00^X^X<86>x\x00^X^X^X`x<86>^^\x00^X^F^Xfæàææ^F~^Xfæàfx<98><98>`^F^X^X~\x00<98>æ<86><9e><9e><9e>ø<9e>þ<9e>~<86><9e>\x00^F^X^F^X^X<98>à^X^^xæx`<98>f<80>^X^X^Xf<80><86>~\x00^X^F^X^X<98>à^X^^x^F<98><80><86>^F^^^Xx\x00^X~\x00<98>æ<86><9e><9e><9e>ø<9e>þ<9e>~<86><9e>\x00^F^X^F^X^X<98>à<98>^^^F^Xx^^æf^F^X^^\x00^Xf<80><86>~\x00^X^F^X^X<98>à<98>^^^F<98>x^F<86>^F^^<98>x\x00^X~\x00<98>æ<86><9e><9e><9e>ø<9e>þ<9e>~<86><9e>\x00^F^X^F^X^X^X^X``<98>^^`<98>f^F^X^X^Xf<80><86>~\x00^X^F^X^X^X^X``<98>^^f<86>^Fø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>ø<86>!îÈó^N<90><9d>^M^HÂ^Bm@[^L1
2021-03-06 20:12:53,810 - octoprint.util.comm - WARNING - The received line contains at least one null byte character at position 110, this hints at some data corruption going on
2021-03-06 20:12:53,821 - octoprint.util.comm - WARNING - Received line:
2021-03-06 20:12:53,821 - octoprint.util.comm - WARNING - | <9d>!\x00<9d>¿MHJ<9d><8a><8a><88><98><8f>N^H Á!Î¨ÞÖ ^H
[... cut ...] B0', 'printer_baudrate': 115200}                                                                                                                                                                                          

### Additional Information

Kickstarter 4.5.2 board, all stock

CR6-SE, CF6-prerelease 4, Octoprint 1.5.3, Raspberry Pi 3 Model B Rev 1.2

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

CF6-Pre4-cr6-se-v4.5.2-mb-2021-02-28-13-49.zip

ritchiedc, 2021-03-07:
I can confirm my printer does the same. I did notice that if Octoprint Baudrate is set to 115200 instead of auto the reboot does not stall and after reboot it connects. It only seems to restart (crash) on first attempt after power on. I'll try reverting to previous versions to see when the problem started.

Sebazzz, 2021-03-07:
These are probably bugs imported from upstream, I suspect either of these two:

- https://github.com/MarlinFirmware/Marlin/issues/21244
- https://github.com/MarlinFirmware/Marlin/issues/21010

I have compiled a release based on the latest sources merged from upstream:

    CF6-Nightly-cr6-se-v4.5.2-mb-2021-03-07-19-11.zip                    CR-6 SE (v4.5.2 motherboard)                                                                    C58098A6476D2D26A116DA63435C16DDDEA9CDE66F566B574EF2FE6DCB3975E6

[CF6-Nightly-cr6-se-v4.5.2-mb-2021-03-07-19-11.zip](https://github.com/CR6Community/Marlin/files/6097708/CF6-Nightly-cr6-se-v4.5.2-mb-2021-03-07-19-11.zip)

Verify that it is flashed in the info menu. If this doesn't work, I suggest you take this upstream.

ritchiedc, 2021-03-07:
The nightly still crashes

gleonard3, 2021-03-07:
just confirmed as well... nightly still crashes as before

Sebazzz, 2021-03-07:
Then I suggest you take the configuration files and report this upstream, probably in one of these issues.

Sorry, this is out of my hands. 👍

closed 2021-03-07 by Sebazzz

Sebazzz, 2021-03-08:
@gleonard3 @ritchiedc One afterthough, can you please try this build:

[CF6-Nightly2-cr6-se-v4.5.2-mb-2021-03-08-10-52.zip](https://github.com/CR6Community/Marlin/files/6100454/CF6-Nightly2-cr6-se-v4.5.2-mb-2021-03-08-10-52.zip)

Verify it is flashed as nightly2 in the info menu.

I have disabled `#define POSTMORTEM_DEBUGGING` in this build.

ritchiedc, 2021-03-08:
It still has problems but does not restart. I use octoprint to monitor prints (mostly video) from my house and often don't remember to start it before starting a print. My prints would sometimes "mysteriously" stop and I'd be back at the home screen. I now know it is related to this issue. My print is no longer interrupted but starting the comm interface is not nearly as clean as with pre3. You may want to mention this in the release notes. I don't know if it is only related to MB 4.5.2. It feels like a pointer is not properly initialized at power-on but is somewhat OK after an internally initiated restart. As a sidenote, once the interface is working I can disconnect and reconnect octoprint with no problems. I may continue to try to trace the source of the problem but I am a novice at git so it may take some time to "undo" changes from pre3 to pre4 to see what change caused the problem.

Sebazzz, 2021-03-08:
Because this is hardware serial communication what goes wrong, I'm pretty sure that this is not my doing but rather something upstream. On the BTT SKR board, which uses software serial, at least for Octoprint, I see no issues.

Sebazzz, 2021-03-09:
Well, this is not true anymore either. I think MEATPACK support is to blame.

Sebazzz, 2021-03-09:
Well, that news makes allows me to sleep at night 😀

[CR6Pre6-Bug212-WPostMortemDbg.zip](https://github.com/CR6Community/Marlin/files/6111204/CR6Pre6-Bug212-WPostMortemDbg.zip) - this build marked with `CR6Pre6-Bug212-WPostMortemDbg` in the info menu has postmortem debugging also enabled. 

If you plan on keeping this installed, here is matching touch screen firmware:
[CR-6-touchscreen-2021-03-09.zip](https://github.com/CR6Community/Marlin/files/6111209/CR-6-touchscreen-2021-03-09.zip)

ritchiedc, 2021-03-09:
I just made a build with the latest commits without Meatpack and with Postmortem and it works without problems. I'll also try your build.

ritchiedc, 2021-03-09:
Just tried your Postmortem build with screen flashed and it works.