CR6Community/Marlin #213: Please consider renaming the "Restore" settings on Screen 70-TuneFWRetract

Thinkersbluff, 2021-03-07:
As discussed under Issue#159, the Retract Override function is designed to allow users to override certain slicer settings during an active print, but some Users may find it difficult to recognize to which specific Slicer settings the override parameters correspond. 
Some Users may also think that the override function is offering functionality that the Slicer does not.

Our User community members use a variety of slicers (Cura, PrusaSlicer, SuperSlicer, IdeaMaker, Slic3r, etc.) and the terms used within those slicers are NOT consistent for the parameters the User can override on this screen.

If Users rely solely upon the labels on this screen to anticipate what each parameter will do, then some Users may think that "Restore", "Length" is the opposite of "Retraction", "Length", rather than the "Extra Prime Distance" from Cura that it actually overrides.  If they do so, they are likely to "mess-up" that print job.

Since the term "Priming" seems to be unique to Cura, changing "Restore" to "Prime" may just change who gets confused...
The term "De-Retraction" seems like a made-up word, but it may also be intuitive for Users to translate into the equivalent setting in whatever slicer they use.
To avoid misinterpreting the "Length" parameter under Restore/Prime/De-Retract - which should probably remain zero in a lot of scenarios - I recommend including that Cura concept of labeling it as an "extra" amount of filament to be added at the end of the full retract/de-retract cycle.

This sketch suggests one alternative arangement for this screen: 

[image]

Sebazzz, 2021-03-08:
Seems like a good suggestion. Will need to wait for someone with Photoshop access. I can't design screens nor work with Photoshop.

Thinkersbluff, 2021-03-09:
So, I figured out how to Photoshop and how to update 23_screens.icl.  Adding Extra to the first label required moving that field down a tad, so I had to edit the touch zone and the data field in DGUS.  I have no clue how to raise a pull request, but I will upload the bmp, psd and icl files here.

Sebazzz, 2021-03-09:
Thanks, looks good.

closed 2021-03-09 by Sebazzz