CR6Community/Marlin #214: Move G28 probing from center to corner

Allan-N, 2021-03-07:
__Issues that do not follow this issue template will be closed__

### Description

To avoid the blob that tends to form at the center of the bed I would like to suggest the probe location be moved to one of the corners.

How might we do this?  If we do not yet have bed leveling info, probe at the center.  But, if we do have leveling info then shouldn't we be able to probe at some other location (e.g. one of the corners) and use the bed leveling info to compute what the value would be at the center?  Moving the probe point to a corner would minimize the center filament blobs.

### Steps to Reproduce

1. Start print
2. G28 probes/confirms the X, Y and Z endstops.  The gotcha is that the Z offset is probed at the center of the bed and tends to leave a blob of filament

**Expected behavior:** 

Ideally, a print can be started without any blobs of filament in the printed area

**Actual behavior:** 

The likelihood of having a blob formed at the center of the bed is high unless one takes extra steps to start the G28 probing at a lower temperature (e.g. @Sebazzz "Start G-code").

### Additional Information

CR-6 SE
4.5.2 mainboard
Rel 6, Pre 4

Sebazzz, 2021-03-07:
> If we do not yet have bed leveling info, probe at the center. But, if we do have leveling info then shouldn't we be able to probe at some other location (e.g. one of the corners) and use the bed leveling info to compute what the value would be at the center?

No, because the bed leveling is determined from the probing position. If you probe in the center, and then later start a print from the side, then all the leveling values are off.

Then there is also the issue that the strain gauge can bend the bed a good bit in the corner, because it has more leverage. That would could issues for users who have their strain gauge set to probe at a higher force. That would make the community firmware instantly unusable.

There is a perfectly good solution (a good start gcode) I don't see the value of probing at the corner. Where you will also leave a blob 😉 You can besides that also argue that you should always probe at the same temperatures.

Allan-N, 2021-03-07:
OK.

closed 2021-03-07 by Allan-N