CR6Community/Marlin #215: [FR] Home without preheating

lukemaurer, 2021-03-10:
__Issues that do not follow this issue template will be closed__

### Description

If I'm doing moves using the menu, I don't usually need them to be 100% precise—it's usually because I'm checking my camera setup. It's then obnoxious to have to wait for it to heat the bed when I tap Home. I'd suggest a pop-up dialog *offering* to preheat if the user taps Home on a cold printer.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** 

[What you expect to happen]

**Actual behavior:** 

[What actually happens]

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

Sebazzz, 2021-03-10:
Thank you for your suggestion.

You may not be aware that in the upcoming Community Firmware release 6 there are settings to customize heating, including not heating at all (set temperatures to zero or ambient). Would that work in your case?

Otherwise I might consider not preheating for homing at all, or at least not waiting for the temperatures from the movement menu. I agree that in the movement menu there should be less need to preheat.

Thinkersbluff, 2021-03-12:
> Otherwise I might consider not preheating for homing at all, or at least not waiting for the temperatures from the movement menu. I agree that in the movement menu there should be less need to preheat.

This sounds like a very good idea.

ssendev, 2021-03-17:
When using the printer as a plotter this is also important but in release 6 preview 5 setting heating to zero solves it.
But it would be nice to have a gcode command to change the settings otherwise i always have to go to the printer.

Sebazzz, 2021-03-17:
There is! Check the output of M503 and look for the line that says C001. Good luck!

Met vriendelijke groet,
Sebastiaan Dammann

lukemaurer, 2021-03-19:
> Thank you for your suggestion.
>
> You may not be aware that in the upcoming Community Firmware release 6 there are settings to customize heating, including not heating at all (set temperatures to zero [...]

Possibly, though I'm always weary of changing a setting that I'll forget to change back. And if I just need to nudge the X-carriage over to fix something, that's a lot of taps if I want to turn heat-on-movement off, do the movement, then turn it back on.

Thinkersbluff, 2021-03-19:
> Possibly, though I'm always weary of changing a setting that I'll forget to change back. And if I just need to nudge the X-carriage over to fix something, that's a lot of taps if I want to turn [...]

Would it help if there were separate Home X and Home Y buttons as options on the Move screen that raised Z a few mm to avoid striking anything but.did not require pre heating before moving?

Sebazzz, 2021-03-20:
This has now been implemented, but only from the movement screen using a trick. I didn't build an extra gcode parameter.

closed 2021-03-20 by Sebazzz