CR6Community/Marlin #218: E-Steps temperature setting not being applied

ilgrank, 2021-03-14:
### Description

While doing e-steps calibration, I've accepted the default of 210°C / 100mm and everything was ok.
Since I was trying to find out if my extruding was skipping at lower temps, I decided to give it another try at 180°C
I set the temp to 180 and tried again. The hotend temp stayed at 210°C all the time, even if the display was still reporting my set temperature of 180°C

### Steps to Reproduce

1. Do an e-steps calibration at 210°C
2. After that, do another calibration at 180°C
3. The temperature remains at 210°C.

**Expected behavior:** 
Hot-End temperature should go to the set temperature.

**Actual behavior:** 
Temperature remains at the default of 210°C
**Logging:**
couldn't capture logs,I'll try tomorrow.

### Additional Information
Stock CR-6, Community FW4 (I've seen CFW5 does not add any fix about this)
4.5.2 motherboard

[Provide pictures or links to videos that clearly demonstrate the issue.]
Same as with the logs, I've reproduced this twice but I did not get pics at the time, I'll try tomorrow.

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6-pre4/CF6-Pre4-cr6-se-v4.5.2-mb-2021-02-28-13-49.zip

Thinkersbluff, 2021-03-14:
Hi.  You are performing esteps calibration, but using a function designed only for lading and unloading filament. By CF 6, we had that figured out and built in an esteps calibration tool.  The CF 4 alpha version will not work for you.

Sebazzz, 2021-03-14:
Are you setting the temperature lower whrn you've already pressed on the "run extrusion" button?

ilgrank, 2021-03-14:
No, I'm setting the temperature first and then "run extrusion"

Thinkersbluff, 2021-03-15:
I can confirm that on my BTT mb with stock TFT, running the 11 March Nightly firmware, if the set temperature is below the current nozzle temperature when Run Extrusion is pressed, the esteps function does not wait for the nozzle to cool and the displayed nozzle target and actual temperatures do remain at their previous (higher than set point temperature) while the esteps extrusion completes.

If the set temperature is higher than the actual temperature, esteps does heat first.

[image]
[image]

Sebazzz, 2021-03-15:
Ah right, yes, I never thought of it that one might want to _cool down_ as well.

commit 02e8aea9 (2021-03-15): E-step calibration: Also cool down if that's what the user asked - fixes #218
closed 2021-03-15 by Sebazzz