CR6Community/Marlin #219: Feature Request Filament feed on change Button

SPCORRIV, 2021-03-14:
Feature Request

Filament feed at pause.

Was wondering if this could be added, recently programmed a filament change at line height on my CR6-SE everthing went as expected to the pause point printer retracted filament for removal. But when you change to the new filament there is no way to feed the extruder IE Load/Unload button this might not be an issue for none Direct drive machines but for all of us who use BMG DD mounted right on top of the hot end it isas you can not self feed  and purge the hot end to the new color.

1. So Slicer pauses this works
2. Slicer retracts filament this works
3. Need feed button to load or unload filament at time of change. (Unlock Extruder Stepper Brake at Pause)
Both Will work to make color changes easier.

Machine CR6-SE
Software Community firmware-20210310-194054.bin on one machine and firmware-20210310-194147.bin on my other machine.

Thank You
Scott 
**

Sebazzz, 2021-03-15:
This is related to https://github.com/MarlinFirmware/Marlin/issues/20591

The problem is that at the current moment we actually don't "know" that a filament change happens. From the UI layer the only think we know is that the print is paused and that the user needs to confirm to continue.

SPCORRIV, 2021-03-16:
Hey Sabazzz

Anyway to add extruder jog button to screen at pause ? Maybe on tune page ?

Thanks

areaknight99, 2021-03-29:
Yes, please add the option to change filament during printing. Load and unload, because sometimes the filament runs out there is no way for a print to continue because the sensor senses that there's no filament so it won't continue. Second the filament already went too deep for the user to pull out manually, so the feature of loading and unloading would be great! Thanks!

Sebazzz, 2021-03-29:
That's a separate thing, please open a separate feature request for that. Thanks!

Met vriendelijke groet,
Sebastiaan Dammann

dkurtycz, 2021-04-03:
I absolutely second this request and will be following.

Thank You

Dan Kurtycz

Sebazzz, 2021-04-12:
Being able to change filament during a print using the existing filament change screen has been implemented via #234

closed 2021-04-12 by Sebazzz