CR6Community/Marlin #220: If Tune from a Paused Print, then Back from Tune Returns User to Screen 37 instead of Screen 39

Thinkersbluff, 2021-03-15:
### Description
If User selects Pause on the Printing Screen, and then selects Tune, when they select Back to return to the Printing (Paused) screen, they are instead returned to the Printing Screen.  There is no "Resume" button, but there is a "Pause" button.  To restart the print, the User has to Pause it again & Confirm, to get back to the Resume button.

### Steps to Reproduce

1. Start a new print
2. Select Pause and Confirm Pause
3. Select Tune
4. Select "Back"
5. User is returned to screen 37 instead of screen 39

**Expected behavior:** 

Back from the Tune screen in this instance should return the user to screen 39.

**Actual behavior:** 

Back instead returns the user to screen 37.

### Additional Information

Using BTT SKR CR6 v1.0 motherboard with stock TFT, running the 11 March Nightly firmware and the btt-w/stock-creality-tft config files.

Current work-around is to select "Pause" again, Confirm Pause again, then use Resume to restart the print.

Sebazzz, 2021-03-15:
I sneaked a possible fix for this in 38dc92b4a7b713c8688f26e498ef5c1a49796ccd - since you are compiling yourself now, can you test that? (Note that the commit involves https://github.com/CR6Community/CR-6-touchscreen/pull/40 so you need to reflash the touch screen as well)

Thinkersbluff, 2021-03-15:
> I sneaked a possible fix for this in [38dc92b](https://github.com/CR6Community/Marlin/commit/38dc92b4a7b713c8688f26e498ef5c1a49796ccd) - since you are compiling yourself now, can you test that? (Note that the commit involves [CR6Community/CR-6-touchscreen#40](https://github.com/CR6Community/CR-6-touchscreen/pull/40) so you need to reflash the touch screen as well)

Sorry - just found this a little while ago.

I tried the latest, but I see it is presently LOCKED according to Discord.
Bug is still present.
Also note that if I select Screen Saver OFF, I get a dotted line box instead of OFF and can not turn it back on again.

commit 165f5345 (2021-03-15): Set status message while pausing print - fixes #220
closed 2021-03-15 by Sebazzz