CR6Community/Marlin #221: [FR] Show what number is being input (temp or axis)

lukemaurer, 2021-03-16:
__Issues that do not follow this issue template will be closed__

### Description

The numeric keypad input doesn't show what it is you're editing. This is especially noticeable on the Move screen, where there are several controls right next to each other, and it's easy to, say, hit X-axis instead of temperature and cause it to attempt a 200mm move out of bounds. A simple “Set Extruder Temp” or “Set X-Axis Position” prompt above the keypad would help a lot.

Sebazzz, 2021-03-16:
I'm not sure DWIN supports something like that. Needs investigation.

Sebazzz, 2021-03-21:
Reading the DWIN documentation, the "Return VP" feature might possibly be used to poke the firmware, and from the firmware we can poke the keyboard.

[image]