CR6Community/Marlin #222: How do i flash the community firmware back to creality?

raulex22, 2021-03-16:
__Issues that do not follow this issue template will be closed__

### Description

im trying to go back to creality firmware, but i can´t, i have tried to do the same process to install the bin files, from the first creality firmware of the cr6 to the most recent, every time i turn on the printer it complete the loading bar and then nothing, it can stay there for hours

### Steps to Reproduce

just insert the sd with any creality firmware

1. [insert creality firmware

**Expected behavior:** 
go back to creality firmware

[What you expect to happen]

**Actual behavior:** 
freeze on loading screen of community firmware

[What actually happens]

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]
First batch of mainboards, got it from kickstarter, no mods

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

really need some help, ive been having a bad time with this for a few days now

Doridian, 2021-03-16:
You need to flash the LCD/TFT as well as the mainboard.
Community FW on mainboard requires community FW on touchscreen.
Same goes the other way around (Creality FW on mainboard requires Creality FW on touchscreen)

Sebazzz, 2021-03-16:
What @Doridian says: the instructions are the same as for the community firmware, but with different files. 

> every time i turn on the printer it complete the loading bar and then nothing

Try renaming the .bin file, reformatting the SD card etc.

closed 2021-03-16 by Sebazzz