CR6Community/Marlin #223: Heating and Homing failure after rollback from Release 5 to official Creality firmware

ryantang30, 2021-03-19:
### Description
After upgrading firmware to Release 5 successfully (was able to home, level bed and print normally), rolling back to any Creality firmware version or Release 3 (As advised in Release 5 release notes) results in nozzle and bed unable to heat up at all, and z-axis homing to always fail (nozzle keeps going down even after opto sensor is blocked and blue light shows up on daughterboard on nozzle assembly. This results in the nozzle crashing into the bed. 

When I re-flash both screen and motherboard firmware back to Release 5, all functions become normal again.

### Steps to Reproduce

1. Flash Release 5 firmware to both motherboard and screen
2. Flash any official Creality or Release 3 screen and motherboard firmware
3. Attempt to auto-home nozzle and heat up nozzle/bed

**Expected behavior:** 

Auto-home completes normally and nozzle/bed heats up normally

**Actual behavior:** 

Nozzle and bed unable to heat up at all, and z-axis homing to always fail (nozzle keeps going down even after opto sensor is blocked and blue light shows up on daughterboard on nozzle assembly. This results in the nozzle crashing into the bed. 

**Logging:**

Did not do any logging as I am not sure how and if its relevant. If required, do let me know.

### Additional Information

V4.5.3 Motherboard
No custom mods

Firmware used for release 5: https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.3-cr6-community-release-5-beta

For rolling back I tried the following firmwares:

Release 3 
Mainboard: https://github.com/CR6Community/Marlin/releases/tag/v2.0.7.1-cr6-community-release-3
Screen: https://github.com/CR6Community/CR-6-touchscreen/releases/tag/v1.0.0-a1

Official Creality V1.0.3.6: Downloaded from the page: https://forums.creality3dofficial.com/download/cr-series/cr-6-se/
Direct File Download: https://secureservercdn.net/192.169.223.13/i6h.45f.myftpupload.com/wp-content/uploads/2020/10/CR-6-SE-32BIT-latest-firmware-V1.0.3.6.rar

Official Creality V2.0.1.3 for V4.5.3 Mainboard: Downloaded from the page: https://forums.creality3dofficial.com/download/cr-series/cr-6-se/
Direct File Download: https://forums.creality3dofficial.com/?smd_process_download=1&download_id=27860

ztakis, 2021-03-19:
Official Creality V1.0.3.6 is for v4.5.2 Mainboard. Have you tried Creality's download page? 
https://www.creality.com/download
The 28 Jan, 2021 releases have a [ 4.5.3mainboard  (C & E version)（common with CR-ERA V1.1.0.3mainboard  -  CR-6SE - V1.0.4.1(1).bin ] that might work with a v4.5.3 Mainboard.

Sebazzz, 2021-03-19:
@ryantang30 Creality incorrectly published v4.5.2 board firmware as v4.5.3 firmware - this has been going for a few months now and apparently they _still_ haven't addressed that. That's the reason the heating doesn't work and homing fails.

Either way, not something that I can fix.

closed 2021-03-19 by Sebazzz