CR6Community/Marlin #228: Immediate reboot of printer when starting a job

ebeng, 2021-03-22:
hi all,

I have to say that this is the only Firmware I got my new Creality CR6 SE got to work. 
MANY MANY thanks for this. The official firmwares didnt work. After I got my printer, I did upgrade from the default 2.0.1.2 which it came with to 2.0.1.3, but everything stopped working until I found this firmware. 

With this firmware I see some issues, when trying to print some stuf, the printer reboots suddenly and it cant resume where it was. When I try to resume, it literallly hits/slices the plate with the nozzle and damages it.

I do have some videos if necessary. 

[Galata_kulesi_v3.zip](https://github.com/CR6Community/Marlin/files/6183379/Galata_kulesi_v3.zip)

this is what I tried to print. Do you guys have also problem with this? Try print it to scale to 10% and set it to Z-level of 0mm. please try print this. with this one the printer immediately stops.

[Galata+Kulesi+(Great+Tower+).zip](https://github.com/CR6Community/Marlin/files/6183414/Galata%2BKulesi%2B.Great%2BTower%2B.zip)

with this one the printer stops in about 5-10 mins after the second layer...

thanks!!!!

Sebazzz, 2021-03-22:
Please use the issue template to fill in all blanks, and reopen the issue. The issue template is there for a reason. Thank you.

closed 2021-03-22 by Sebazzz

ritchiedc, 2021-03-22:
Probably not your problem, but I've had issues with static discharges when I touch the printer causing it to reset.