CR6Community/Marlin #231: Neopixel support on DGUS

Ranney1, 2021-03-28:
Sorry, had to ask...

Would be great to have neopixel support on the DGUS!

Options:

- LED button can (also) turn Neopixel on/off
- Special page on DGUS with color/brightness control
- Errors (mismatch firmware) can control neopixel with red/yellow/blue state. (like events as implemented in Marlin).

Is this possible? A lot of work? Of course, if I'm the only one that is using neopixels than please remove this request!

Sebazzz, 2021-03-28:
Moving to Marlin.

Sebazzz, 2021-03-28:
Sounds like fun, first need to buy another neopixel for my CR-6 with stock touch screen 😉

Sebazzz, 2021-04-11:
I got [my neopixel installed now](https://damsteen.nl/blog/2021/04/11/how-to-mount-neopixel-to-creality-cr6). Any suggestion where to place this in the menu @Ranney1? Also taking #121 into account of course.

Sebazzz, 2021-04-11:
Alright. This is implemented using something we didn't use before: slider controls.

Hopefully @ztakis / @grobux have time to make something decent of it, instead of my dabbling in Paint.NET 😄

Implementation notes:

- Added page 81 RGB
- Added icon 30, 31, 32, 33, 34 which represent the slider knob itself for R, G, B, W, and brightness respectively. 
- Added navigation in the misc settings (until we're doing something serious with #121), and the extra tuning menu

Sebazzz, 2021-04-12:
@ztakis I made the RGB button into its own icon, so it only shows when applicable. You may want to update the PSD, but regeneration of the bitmaps is not needed 😉

This feature request has been fully implemented now.

closed 2021-04-12 by Sebazzz