CR6Community/Marlin #232: No filament change during printing (Load/Unload)

areaknight99, 2021-03-29:
Printer cr-6 se.
I tested the filament run out sensor and it work like it should. The problem is when it detective that there is no filament, the print paused. There is no option for loading or unloading the filament. This is an issue because sometimes the filament runs out and there is no way for a print to continue because the sensor senses that there's no filament so it won't continue. (Even if you press continue, it will just pause again) Second issue is that the filament already went too deep for the user to pull out manually, so the feature of loading and unloading when filament runs out would be great! Thanks!

Thinkersbluff, 2021-03-30:
@areaknight99 - In my experience, if I am printing when the filament runs out, the runout sensor stops the extruder before the remnant has left the extruder gears.  Marlin retracts the remainder when the head has parked, as part of the M600 command sequence, so that I can pull it out and replace it.  

I have never tried to do it, so I don't know - if you do try to Continue a print without first loading a new filament, does that push the remnant back in?

JamesSF69, 2021-03-30:
Of course right after this was posted, I ran into this issue. There is no way to get to a screen that allows a "change filament" action. When the actual printer runs into the filament sensor detecting no filament.
No way to "unload/reload the filament.  And the filament is in the extruder and no way to pull it out. (baring disassembling the extruder)
Can the M701/2 gcode be used when the printer is "waiting" to have the filament refreshed (loaded, unloaded) i.e. Using a custom "send code" action from Octoprint to "retract/unload or feed/load" filament?

areaknight99, 2021-03-30:
> Of course right after this was posted, I ran into this issue. There is no way to get to a screen that allows a "change filament" action. When the actual printer runs into the filament sensor [...]

Yes! Someone with the same issues! Now I know more people are going to run into this problem, so we need this feature for the upcoming firmware update! It would be great!

rockymtnpatriot, 2021-03-30:
I have only had this happen once and it was on version Pre5. As I recall I
either had a button on the TFT or in Octoprint to "Resume" after I fixed
the filament break and was able to re-start the print. I have the
InlineConfirm plugin to Octoprint which may have been where the message
came through, I just don't recall what I saw it on. Do you have that plugin?

JamesSF69, 2021-03-30:
> I have only had this happen once and it was on version Pre5. As I recall I either had a button on the TFT or in Octoprint to "Resume" after I fixed the filament break and was able to re-start the [...]
Resuming the print is not the issue, that is fine. But I have no way to change the filament besides physically pulling it out and pushing it back in, or in my case stand there physically feeding in the new filament until it reached the gears of the extruder.
It would be nicer to have the "unload/load" filament menu handy when the filament sensor alerts that there is no filament.

Thinkersbluff, 2021-03-31:
@areaknight99 - Can you provide more of the information required per the Issues Template, please?

[What is your hardware configuration? Which mainboard? Custom mods like different hot-ends and extruders? BTT touch screen?]

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

Are you printing from SD card or from a Host (like Octoprint)?

Do you have a serial log of the messages you get when the filament sensor detects a runout?

There should be an M600 command when the filament sensor detects a runout.

Configuration.h is programmed with this setting as of Release 5:
  #define FILAMENT_RUNOUT_SCRIPT "M600"

The M600 command is programmed to eject the residual filament automatically, [per this document from Marlin](https://marlinfw.org/docs/gcode/M600.html):
https://marlinfw.org/docs/gcode/M600.html

If that is not happening on your machine, then maybe M600 is not being issued?

s3frank, 2021-04-01:
I can confirm this issue last night as well. Ran out, print just "finished" and that was that.
I had this on the original firmware once and it paused, allowed me to reload and continue.

Thinkersbluff, 2021-04-01:
Sorry, I cannot reproduce this problem on Release 6-Final, with the BTT board & stock display.

I get these results whether printing from SD card or from Octoprint:
- Wait for printing of model to start
- Cut filament at input to runout sensor & wait for system to detect the break
- Sensor light goes out
- Print head moves to the Park position
- Extruder unloads a length of the remnant (which includes purging a few additional mm first)
- I pull the rest of it out
- Printer is beeping
- I load a new filament
- I pull the purged amount from the nozzle with tweezers
- the nozzle & bed are still at printing temps
- I push Resume and the print resumes

I am using the metal dual gear extruder, not stock, but the filament runout sensor and extruder are still mounted in their stock positions.

s3frank, 2021-04-01:
I will upgrade next week to final build and try. Thanks for testing!

Sebazzz, 2021-04-12:
Feel free to reopen if this issue is still there.

closed 2021-04-12 by Sebazzz