CR6Community/Marlin #234: Pause at Height with Cura/Octoprint/Community 6

johnelle, 2021-04-01:
__Issues that do not follow this issue template will be closed__

### Description

Information request.  I have not successfully implemented Pause to change the filament color.  Screen is locked in print mode so I can't manipulate the feed or do any movement.  Just using the default Cura with no settings or gcode.  

### Steps to Reproduce

Just add default "pause at height" to my slice.

**Expected behavior:** 

Pause at both the Octoprint and screen.  Allow navigation to extruder and moves.   On resume the machine rehomes and continues printing on top of the last layer.

**Actual behavior:** 

Pause on both ends but machine only allows access to print screen.   Can only do a manual filament change.

**Logging:**

### Additional Information

4.2 stock MB

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
Community 6

Sebazzz, 2021-04-01:
This is expected. In printing you can't use the filament change screen (neither can you in stock Marlin).

To do filament change, [check this resources](https://damsteen.nl/blog/2020/12/26/how-to-do-filament-change-with-cr6-community-firmware-release-5). Otherwise, check issue #232

Sebazzz, 2021-04-12:
Filament change screen is now available during a print. Needed some rework, but I got it done.

@ztakis I've added the filament change button from the prepare screen to the extra tuning screen.

closed 2021-04-12 by Sebazzz