CR6Community/Marlin #235: [FR] Terminal Interface

defiant07, 2021-04-01:
It would be cool if a terminal interface like the one found in BIQU/BigTreeTech's firmware could be added so the user could issue gcode commands directly from the printer, thereby, eliminating the need for additional hardware and software...wish I had sufficient skills to help y'all...many thanks for all your efforts!

Sebazzz, 2021-04-01:
That'd be very difficult probably. Marlin doesn't allow catching serial output, which we'd then need to send to the display. So that feature would probably be input only, so you never see the output of any command. That limits it to things like e-step calibration and PID tuning, both of which have already an excellent user interface.

defiant07, 2021-04-02:
Ahh, interesting, I understand, not being able to hook the output would definitely limit it's utility and I imagine making and integrating a hooking framework would not be feasible/worth the effort. Thanks again, keep up the awesome work!

Ranney1, 2021-04-06:
On the BTT screen is a complete Terminal screen incl. response. Maybe that the implementation for that screen can be used/helpful?

defiant07, 2021-04-07:
Idk, I've only seen it on youtube videos, but I intend to get BIQU BX once it becomes available again.  Yeah, porting it could be viable, but that's beyond my knowledge :/

Sebazzz, 2021-04-07:
Porting is not viable, it is totally different technology.

defiant07, 2021-04-08:
Sorry, guess I've sufficiently demonstrated my lack of knowledge, lol.