CR6Community/Marlin #236: Feature Request: Filament Wipe After Change Filament

johnelle, 2021-04-03:
__Issues that do not follow this issue template will be closed__

### Description

Even if the nozzle is clean after I swap filaments the first extrusion is a "blob" but then things continue as normal.  I conclude from this that a wipe like the one done on print start is needed for a smooth transition.

*Description of the bug or requested feature*### Steps to Reproduce

After the final reheat and extrusion of a new filament the machine extrudes a 100mm or so on the table before resuming the print.

**Expected behavior:** 

[What you expect to happen]

**Actual behavior:** 

[What actually happens]

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

Running Community 6 on stock hardware.

[If you compiled the firmware yourself - Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files]

[Provide pictures or links to videos that clearly demonstrate the issue.]

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
Community 6.

Sebazzz, 2021-04-03:
Sorry, that [is not possible at the moment](https://github.com/MarlinFirmware/Marlin/issues/20591). Marlin doesn't have the hooks to allow filament purge from the touch screen interfacing code. You need to fix this in your process, for now.

closed 2021-04-03 by Sebazzz

johnelle, 2021-04-03:
Would love to "fix it in my process"  How?

Sebazzz, 2021-04-03:
Specify no purge length for the M600 command, and when you've fed filament, give it some time for the filament to droop out before resuming the print. Optionally, but in my experience not necessary, you can have something ready to wipe the nozzle with.

Z3R0-89, 2021-07-29:
> 
> Would love to "fix it in my process" How?
> […](#)
> -- John Ellenberger Groton, MA

My workaround so far is to change the flow rate to 1% just before the purge ends and once the nozzle starts printing again and touches the part, set it back to 100%. This way, the blob is quite small.