CR6Community/Marlin #239: Possible to add support for Laser module?

ScarletStar, 2021-04-06:
### Description

Since this is more of a niche use case, it's kind of low priority, but I figured I'd just put it here anyways. 
I haven't looked into this deeply, but it seems like there is a FAN2 header that is not populated at least on my 4.5.2 board, so I'm not sure if it is routed to its' own IO pin for independent control. But if it is, would it be possible to add a laser engraving mode for use with Creality's laser module (or any other 12V/24V laser module for that matter)?
https://www.creality3dofficial.com/products/creality-blue-violet-light

This would save the user a lot of fiddling around with settings that aren't even named properly for engraving. Settings that we don't even have right now I think. As in turning on and off the blower fan and turning off z Axis movement. In other words the official way of using the official module is basically a hack, and it shouldn't be.

### Suggested behavior:

1. Activate Laser Engraving Mode 
2. Z-Axis goes up to 100mm or so and stays fixed (maybe make the fixed Z Position a setting)
3. Button to activate/deactivate laser becomes available (for manual calibration of the focal spot)
4. Engrave Button -> file select menu (just like normal print mode)
5. (Maybe check file for any z-Axis movement which shouldn't be there to issue a warning message?)

### Additional Information

Not sure if this could be it's own tile since we probably don't want any scrolling on that screen...Or just a sub Mode of the Printing Tile? It should probably be hidden by default since most people probably don't have a laser module. Maybe with a switch in the settings to unhide it.

Ranney1, 2021-04-07:
As far as i know, on the 4.5.2 board this is an 'always on/24v' port. Only on the BTT board there is an free FAN port that is controllable (together with the mainboard FAN) I think that it is not a good idea to use this FAN port because this could result in an mainboard (drivers) that is getting to hot (laser off is mainboard fan off).

Maybe that you can make something in the hotend because there is an FAN port (for the part cooling) that is controllable?

Sebazzz, 2021-04-07:
@Ranney1 is correct- the ports are linked to 24V direct, with no MOSFET or anything in between. On the BTT board it is controlled through the mosfet.

Sebazzz, 2021-04-07:
Since the laser is hardwired (and due to the way PWM is implement on the board apparently cannot be controlled properly even) this is not something I can be implemented I'm afraid. Feel free to re-open if I'm wrong.

closed 2021-04-07 by Sebazzz