CR6Community/Marlin #240: feature request: no "z-hopping" when filament load/unload and not print in progress

schirrm, 2021-04-08:
### Description

When doing a filament load / unload the printer does move the z-axis up/down. 
As the menu item was used in a not-printing-situation, I was wondering what that's good for. Whilst printing I'd expext the hotend to go the far left end and a bit up for waiting for filament change.
When not printing I'd prefer the printer not to move the z-axis

### Steps to Reproduce

well as not really a bug, not required.

1. turn printer on
2. go to load / unload filament
3. wait for heating up

**Expected behavior:** 

after heating up the filament should be load / unloaded immidiately. with no up / down of the z-axis

**Actual behavior:** 

after heating up the z-axis is moved upwards, lowered afters loading

**Logging:**
no log

### Additional Information

creality .2
[precompiled]
[don't think they are required here]
[cr6 se community, release 6]

Thinkersbluff, 2021-04-08:
Marlin is designed to behave this way. The purpose is to ensure that the nozzle is away from the bed before extruding ( even retraction extruded first, to clear the bulb shape off the filament before pulling it back)

schirrm, 2021-04-08:
thanks for the quick reply. understood.
just because I'm lazy ;) is Z<distance> taken from a configuration or "hard coded"? maybe something for a custom build . . .

Sebazzz, 2021-04-11:
This is intentional, as @Thinkersbluff mentioned. You can change `NOZZLE_PARK_POINT` in `Configuration.h` and [use this guide to compile the firmware](https://damsteen.nl/blog/2021/01/08/how-to-compile-cr6community-marlin-with-vscode-platformio). I do not recommend it though: it will also affect the pause behaviour and filament change.

closed 2021-04-11 by Sebazzz