CR6Community/Marlin #241: Community Firmware 6.1 - Provisional release notes

Sebazzz, 2021-04-13:
**This release is a smaller release to implement some feature requests and issues reported in release 6. All the testing and feedback has made this firmware better and better. Thank you for testing and sending your feedback.**

**TLDR: Added PID tuning, interactive e-steps calibration, overriding slicer retractions, custom ABL probing settings, many more configuration knobs and stability improvements.**

## Table of contents

- Hardware Supported
- Important Safety Warning
- Community Firmware Architecture
- Hardware Dependencies and this Community Firmware
- Previous Community Firmware Release History
- New Features at Release 6
- Comparisons with Creality stock firmware
- Bug Fixes at Release 6
- Known issues at Release 6
- Flashing instructions
- Files listing
- FAQ
- How to Report issues
- How to Contribute

## Hardware Supported

This release of the CR-6 Community Firmware supports the following hardware configurations:

- Creality CR-6 SE with either a v4.5.2, v4.5.3 or or v1.1-ERA motherboard. See the [FAQ](#faq) section to know which board you have.
- Creality CR-6 SE with [BigTreeTech (BTT) SKR CR-6 board](https://damsteen.nl/blog/2021/01/24/bigtreetrech-skr-cr6-review) and:
   - a BTT Thin Film Transistor (TFT) v3.0 touch screen
OR - a Stock Creality TFT touch screen
- Creality CR-6 MAX with v4.5.3 motherboard
- Creality CR-6 MAX with BigTreeTech SKR CR-6 board and:
   - a BTT TFT v3.0 touch screen
OR - a Stock Creality TFT touch screen

# Important Safety Warnings
1) **This release does NOT prevent users from setting the Nozzle Temperature higher than 260C.**

>Marlin normally refuses to execute commands like M303 (PID) or to allow users to enter a heater setting that exceeds the safe maximum for the nozzle, bed or chamber or laser cooler.  **That feature is not working in this release for nozzle temperatures.**  It does work for bed temperatures. We don't know whether it is working for the chamber or laser cooler.

2) **The Community Firmware enables the watchdog timer function for STM32F1 processors - if Marlin would freeze from some reason (like due to SD card issues) the printer is automatically reset and the heaters are disabled - this feature is not enabled on the stock Creality or BTT firmware**

Included in the downloadable Assets on this page is one pre-compiled version of the Community Firmware with this Watchdog Timer disabled.
  
CR6Comm-CF6-Final-cr6-se-v4.5.2-mb-no-watchdog-2021-03-27-15-53.zip is provided to support the troubleshooting by owners of printers using the Community Firmware on 4.5.2 motherboards, which appear to be repeatedly rebooting themselves.
  
**This special version of the firmware is designed only to help users confirm or eliminate the possibility that the watchdog timer function is rebooting their machine.  It is not intended to be used on unattended printers or to "solve" the rebooting problem.**

## Community Firmware Architecture

**This Community Firmware is comprised of two firmwares to flash:**

 1. A Community Firmware Motherboard module, based on Marlin bugfix 2.0
 2. A Community Firmware Display module (for the Creality Stock Displays only).
  
The Marlin firmware on which this Community Firmware is based is designed to facilitate the customization of the firmware to run on a wide variety of hardware configurations.

Compiled versions of the Motherboard firmware are provided in the Assets section at the bottom of this page, each uniquely configured for one of the motherboards listed under Supported Hardware, above.
The latest version of the Community Firmware Motherboard module is maintained on [the CR6Community/Marlin GitHub page.](https://github.com/CR6Community/CR-6-touchscreen)

### The Community Firmware Display module is ONLY required on Creality stock (DWIN) TFT displays.  

A compiled version of the Display firmware module + the DGUS OS installer program are included as an embedded zip file inside the Motherboard firmware zip files, for each of the supported hardware configurations that include the stock Creality TFT. 
The latest version of the Community Firmware Display module is maintained on [the CR6Community/CR-6-touchscreen GitHub page.](https://github.com/CR6Community/CR-6-touchscreen)

**The Display firmware module and the Motherboard firmware module on printers using both Creality Motherboard hardware and Creality TFT hardware must be a matched-pair.** 

On printers using both Creality Motherboard hardware and Creality TFT hardware, Community Firmware UI controls and displays are implemented as a collaboration between the Display firmware and the Motherboard firmware.

e.g.: On the Creality TFT: Pressing a "button" on the Creality TFT results in the Community Firmware Display module sending messages to the Community Firmware Motherboard module, sometimes eliciting a response back.  
When the Community Firmware Motherboard module needs to display information to the user, it sends a message to the Community Firmware Display module, which then displays the information on the TFT screen.
[... 292 lines cut ...]

```
File name                                                                          Description                                                                      SHA256 hash
---------                                                                          -----------                                                                      -----------

```

#### _For use with the CR6-MAX:_

```
File name                                                                          Description                                                                      SHA256 hash
---------                                                                          -----------                                                                      -----------

```

#### _Source Files:_

```
File name                                                                          Description                                                                      
---------                                                                          -----------                                                                      
Source Files (zip)                                                                 The source files for the motherboard firmware in this release (.zip file)
Source File (tar.gz)                                                               The source files for the motherboard firmware in this release, (.tar.gz file)
```

__Downloads below - find the correct download for your printer(s) in the file listing above__

Thinkersbluff, 2021-04-13:
Could this just summarize what changed at 6.1, please?
“ TLDR: Added PID tuning, interactive e-steps calibration, overriding slicer retractions, custom ABL probing settings, many more configuration knobs and stability improvements.”

Thinkersbluff, 2021-04-13:
Should Issue#237 be listed under milestone 6.1?

Sebazzz, 2021-04-13:
> 
> 
> Should Issue#237 be listed under milestone 6.1?

It is, isn't it?

Thinkersbluff, 2021-04-13:
> > Should Issue#237 be listed under milestone 6.1?
> 
> It is, isn't it?

Am I looking in the wrong place?
[image]

Sebazzz, 2021-04-13:
> 
> 
> > > Should Issue#237 be listed under milestone 6.1?
> > 
> > 
> > It is, isn't it?
> 
> Am I looking in the wrong place?
> [image]

That issue is closed, isn't it? You're filtering on open issues.

Thinkersbluff, 2021-04-13:
Hah! 
Mind if I update the Release Notes to add that Filter setting where it says,"to view a complete list.."?  Seems we don't see a complete list of closed unless we also filter on closed...

Sebazzz, 2021-04-13:
> 
> 
> Hah!
> Mind if I update the Release Notes to add that Filter setting where it says,"to view a complete list.."? Seems we don't see a complete list of closed unless we also filter on closed...

Once it is actually released then all the issues will normally be in a closed state as well, so then it doesn't matter.

Thinkersbluff, 2021-04-13:
For those of us not as familiar with navigating in GitHub, please rephrase:
"For a full list of added features and bugs fixed at Release 6.1, see the Community Firmware Release 6.1 milestone.
For a full list of added features and bugs fixed at Release 6, see the Community Firmware Release 6 milestone."

To read:
"For a full list of added features and bugs fixed at Release 6.1, see the Community Firmware Release 6.1 milestone and select "Closed".
For a full list of added features and bugs fixed at Release 6, see the Community Firmware Release 6 milestone and select "Closed".

Sebazzz, 2021-04-13:
That is not necessary, like I said: when CF6.1 is released, the "open" filter is not preselected, hence all items are visible 😉

Thinkersbluff, 2021-04-16:
This particular fix for end-stops not triggering on Creality MB looks important for CF 6.1 users to know about. 
https://github.com/CR6Community/Marlin/commit/6ea5556c563a4d57c35ed4cf362b39fe984b7615

Would it help to put a corresponding issue on file, so that it shows up when they hit that link for a complete list of all issues resolved?

BloodyRain2k, 2021-05-06:
Does 6.1 fix the M300 command? Or is it even a bug in CF 6.0 that my CR-6 only produces a single beep, no matter the parameters I use for M300?

It's still way better than stock: knowing when it finished preparing and starts to print, so I can check that the first few layers are done properly, and knowing when it's done.

But I'm still wondering why it just has a single kind of beep, of if my CR-6 is just "special".

Sebazzz, 2021-05-06:
No bug has been reported regarding M300.

closed 2021-06-04 by Sebazzz