CR6Community/Marlin #244: Suggestion - homing bed touch location

Playfutbol2, 2021-04-16:
Instead of touching the center of the bed, what if the homing bed touch took place at one of the 16 leveling locations? If it touched the front left location (the 1st bed level point), that would give 2 benefits. First, it would leave a blob at a location less likely to be printed over. Second, it would link that homing data point to an actual data point from the bed mesh (instead of an interpolation between the center 4 points), which might make it slightly more accurate. Sorry if I didn't explain this well, and feel free to reach out for clarification if needed.

Sebazzz, 2021-04-18:
I have put some thought into this, but this probably isn't a good idea, check issue #214 for the rationale. 

Instead of moving to the side, I'd recommend [this start gcode](https://damsteen.nl/blog/2021/02/11/creality-cr6-community-firmware-start-print-without-drooping) instead.

Playfutbol2, 2021-04-19:
Cool, that makes sense! Thanks for all your work on this firmware, it makes
the printer so much better!

closed 2021-04-19 by Sebazzz