CR6Community/Marlin #245: Speaker constantly beeping following filament change

cghildreth, 2021-04-17:
### Description

BTT TFT 3.5 Screen speaker constant beeping after filament change mid print

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

1. Using Cura 4.8, add a Filament Change script using the post-processing menu (park position set to X10, Y10).  
2. Slice the file using Cura.  
3. Upload the model to OctoPrint and start the job.
4. Printer behaves perfectly through the end of the layer immediately before the pause is inserted.  
5. The printhead travels to its parked position and pauses.  Filament retracts the expected distance (20mm, set using Cura).  After a minute or two, the dialogue on the TFT3.5 screen pops up to indicate that the heater timed out, and gives you the option to reheat.  This appears to only disable the hotend heater, as the bed maintains 60 degree temp.  
6. I select reheat to heat the hot end back up.  (note that Step 10 occurs regardless of whether or not a reheat is required).  Then I manually insert the filament and manually push filament to purge.  
7. Then I select continue.  The printer then automatically purges, and gives me the option to purge more if I want.  I've done this both ways where I pick "Purge more" or just immediately select "Continue".  
8. If I select "Purge More", then the printer appears to purge more just the way I would expect.  
9. When I select "Continue" (either immediately or after selecting "Purge More"), the printer resumes printing accurately.  
10. However, within 30 seconds (timing seems to vary slightly), the speaker on the TFT3.5 starts beeping rapidly (seems like about .5 seconds on, .25 seconds off, repeating continuously), and will not stop regardless of what I do.  The TFT3.5 screen also reacts much more slowly to presses than normal.  The only way I can get the beeping to stop is to reset the printer.  

**Expected behavior:** 
I expect steps 1-9 to happen.  Step 10 is unexpected.  

**Actual behavior:** 

See Step 10.  

### Additional Information

CR-6 Max, SKR CR6 board, TFT3.5 screen, EZR Struder, Pi 4B running octoprint.  Printer is running Community Firmware 6.1.  The Screen is running the up-to-date firmware from BTT.  No other mods.  

Firmware used was the pre-compiled firmware.bin from: https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6.1-pre1/CF6.1-Pre1-cr6-max-btt-skr-cr6-with-btt-tft-2021-04-14-20-37.zip

Sebazzz, 2021-04-18:
To be clear, this happens on the BTT TFT, not the stock touch screen?

Do you have a minimal reproducible gcode file, so as short as possible?

cghildreth, 2021-04-19:
My issue is occurring with the BTT TFT.  I unfortunately no longer have the stock touch screen to be able to test.  One other item that I noticed but hadn't mentioned in my original post - the BTT TFT is locked into the print status menu.  

I also noted yesterday that the same issue occurs when the filament runout sensor gets tripped - my assumption is that the filament sensor trip issues a M600 command?  

BTW, my workaround for the moment is to just turn the speaker off.  The print will complete accurately so long as there are no other interrupts.  A second filament change in the same print doesn't appear to be possible - the printer just stops when the second M600 command is issued.  

Minimal gcode file is attached.  
[CCR6SE_Minimal Cube for Test.txt](https://github.com/CR6Community/Marlin/files/6336338/CCR6SE_Minimal.Cube.for.Test.txt)

Sebazzz, 2021-04-20:
@cghildreth Do you have the issue without Octoprint as well? Do you use touch mode or MarlinUI (so with the rotary encoder)?

cghildreth, 2021-04-21:
I am using touch mode.  I'll test without octoprint and report back.

cghildreth, 2021-04-21:
@Sebazzz The minimal cube gcode that I uploaded above printed perfectly from the TFT SD Card slot - none of the unanticipated behaviours that I had starting the print from OctoPrint.   I couldn't get the print to even start from the SD slot on the SKR CR6 - I got a heating error of all things.

Sebazzz, 2021-04-21:
Okay, so the issue occurs when printing from Octoprint while the TFT is in touch mode?

cghildreth, 2021-04-23:
Sorry for the slow response.  The last couple of days have been crazy with work.  I probably won't be able to test more until tomorrow, but I will get back to you ASAP.

Sebazzz, 2021-04-23:
The beep when TFT mode is active is actually not an bug, or not a Marlin/firmware bug at least. When a filament change happens, Marlin signals over all serial ports that it is waiting for the user. It does it in quite a frequency. 

The BTT TFT is just lagging behind, catching up on the beeping.

closed 2021-05-01 by cghildreth