CR6Community/Marlin #247: [FR] Endstop test menu

esunayg, 2021-04-17:
It would be nice to check if they are working and mainboard gets the signal via the LCD. And possibly after flash first boot it should ask user to push endstops by hand one by one before powering any stepper.

Sebazzz, 2021-04-18:
The problem is that there isn't any first boot defined. There is the reset to factory settings of course, but we also have to consider the complexity that comes with half flashed screen or motherboard firmwares etc. For this reason I'd like to remain that part of the firmware as-is: boot to the main menu.

That said, this diagnostics screen would be very useful, given the pain the CR-6 users can experience.

esunayg, 2021-04-18:
If it would have the filament sensor and the strain gauge (I know led tells that but it doesnt make sure mb gets it) would be perfect as any other cf upgrade were. :)

ztakis, 2021-04-18:
> The problem is that there isn't any first boot defined.

Can you imagine an equivalent to "safe boot diagnostics" by quickly touching and holding during the loading bar?

BloodyRain2k, 2023-04-02:
Regarding the endstops did I just discover a "bug" with the firmware with my CR6-SE. At the very least I consider it a bug.

When the UI menu for moving the axes is used and you accidentally make the printer drive the Y axis into the endstop, then it does NOT stop.

I just finished a print and was trying to take a look at the duct on my print head and wanted the bed out of the way.
Since I already was in the UI for getting the Z axis up near the end, I told the bed to go backwards.

But since the printer disabled the drivers after printing and me having nudged it a bit, I accidentally told it to go further back than it physically could. And instead of noticing the endstop signal, the printer just kept going and torturing the motor and pulley : (

I'm so glad that apparently nothing got damage, but it make me panic and cut the power, because there wasn't an abort button for the move command either...

At the VERY LEAST would I expect the printer to STOP going after it hits an endstop.

Thinkersbluff, 2023-04-02:
How are you commanding the movement?  
Dialing-in a destination? Commanding a random number of incremental steps?

Have you tried manually testing the endstops with gcode M119? https://marlinfw.org/docs/gcode/M119.html

ritchiedc, 2023-04-02:
@BloodyRain2k For the sake of this discussion I'm going to assume you tried to move the bed after a print was complete and the steppers had been disabled. In this case what you see as a bug is intended operation.

Once the steppers have been disabled Marlin no longer knows where the nozzle is (x, y or z). The only way for it to know is to re-enable the steppers and home, but it cannot home if there is a printed part on the bed. Disabling movement without homing would effectively disallow remote movement via octoprint or similar which I use on a regular basis.

Having a move monitor the endstops only helps in one direction. For example, the x endstop is only activated when the nozzle is moved all the way to the left on the CR-6. You would still not see the endstop going all the way to the right. To be "safe" allowing only left moves would be impractical. Y and z would have similar limitations.

You could require that a home be commanded before moving the nozzle via the GUI but most of us are too impatient to do that and this would disallow remote movement as mentioned above. So the compromise is to allow the moves knowing that the user could command it past its travel limits but leave it up to the user to know what they are doing. You are correct that the steppers make a terrible racket when trying to move past the end of travel. Believe it or not it is not hurting the printer, at least in x and y. One could make the argument that z could be a problem with the CR-6 given the strain gauge. That sound is made by the stepper motors skipping as they are unable to rotate and not the gear grinding on the belt. But I too hate the sound and turn off the printer when I make the mistake of commanding the steppers beyond their travel limits.

If I misunderstood your issue please disregard the above.

BloodyRain2k, 2023-04-03:
@Thinkersbluff the numbers were red and I am aware that that indicates that Marlin turned off the steppers and with that also lost the current position.
And as I said, I used the UI menu for moving the axis in the firmware, where the steps can be chosen from 0.1mm to 10mm.
I was stepping the Y axis back in 10mm steps and I wasn't careful with that, because I did NOT expect the printer to completely ignore itself running into the end-stop that's at the end of the Y axis.

@ritchiedc I am also aware that monitoring the endstops would only prevent the printer from crashing any axis in one direction, because it is literally unable to tell when it's driving into the end of the other.

And my end-stops are not broken, or at least they weren't before this. I can only hope that the Y axis one didn't take damage.
I ran into this after running a print, and if they were broken or otherwise off, then I'd have noticed that already when it homed itself in the beginning.

What I am seeing as a bug, and anyone with better understanding of printer firmware can correct me on that, is that the printer *ignores* an end-stop signal as it's running into and past it.
It obviously can only prevent that, if end-stops are present and only in the directions where they are present; which in the CR6's case are X left, Y back and Z down.
So I see it as a bug that it runs the bed back into the Y end-stop and still keeps going, just because I accidentally told it to.

Thinkersbluff, 2023-04-03:
@BloodyRain2k : I understand your perspective.
At this release, that is not a bug, that is a feature.

When the motors are disabled, you can physically push the bed to the back and push the gantry to the side. You do not need to move them with motors.  You can also manually raise and lower the Z axis.

Whether those motors were disabled manually or automatically, the result is the same.  The numbers turn red and Marlin has no idea where the beds are, anymore, until you run G28.